m-labs/nmigen #1: Example of embedding exisiting Verilog?

mithro, 2018-12-13:
Could we get an example of embedding existing Verilog code? With specials having gone, it's unclear to me how that works?

jordens, 2018-12-13:
`Instance`?

whitequark, 2018-12-14:
Yeah. There are no `Instance`s right now (not implemented), and I'm still thinking about the best way to handle them.

whitequark, 2018-12-17:
Added the example:
```python
from nmigen import *
from nmigen.back import rtlil, verilog

class System:
    def __init__(self):
        self.adr   = Signal(16)
        self.dat_r = Signal(8)
        self.dat_w = Signal(8)
        self.we    = Signal()

    def get_fragment(self, platform):
        m = Module()
        m.submodules += Instance("CPU",
            p_RESET_ADDR=0xfff0,
            i_d_adr  =self.adr,
            i_d_dat_r=self.dat_r,
            o_d_dat_w=self.dat_w,
        )
        return m.lower(platform)

sys  = System()
frag = sys.get_fragment(platform=None)
# print(rtlil.convert(frag, ports=[sys.adr, sys.dat_r, sys.dat_w, sys.we]))
print(verilog.convert(frag, ports=[sys.adr, sys.dat_r, sys.dat_w, sys.we]))
```

```verilog
/* Generated by Yosys 0.8+90 (git sha1 ec3698a8, clang 6.0.1-9.2 -fPIC -Os) */

(* top =  1  *)
(* generator = "nMigen" *)
module top(dat_r, dat_w, adr);
  (* src = "examples/inst.py:7" *)
  input [15:0] adr;
  (* src = "examples/inst.py:8" *)
  input [7:0] dat_r;
  (* src = "examples/inst.py:9" *)
  output [7:0] dat_w;
  CPU #(
    .RESET_ADDR(32'd65520)
  ) CPU (
    .d_adr(adr),
    .d_dat_r(dat_r),
    .d_dat_w(dat_w)
  );
endmodule
```

closed 2018-12-17 by whitequark