m-labs/nmigen #2: How do multiple If statements work?

mithro, 2018-12-14:
I'm interested in how the `If` statement actually works. The `m` module seems to track the current "state" in some way?

```python
        with m.If(self.a):
            m.d.comb += self.o.eq(self.sub.o)
        with m.If(self.b):
            m.d.comb += self.o.eq(self.add.o)
        with m.Else():
            m.d.comb += self.o.eq(self.rnd.o)
        return m.lower(platform)
```

How does it actually know that the `m.Else()` applies to the first `m.If(`?

whitequark, 2018-12-14:
> How does it actually know that the `m.Else()` applies to the first `m.If(`?

It doesn't, just like it wouldn't in the following Python code:
```
if a:
  o = sub.o
if b:
  o = add.o
else:
  o = rnd.o
```
The `m.Else()` applies to the *second* `m.If()`.

closed 2018-12-14 by whitequark

mithro, 2018-12-14:
I actually meant the second `m.If()`. Is m tracking what the current if is?

whitequark, 2018-12-14:
Ah. Yes. It is tracking the current nesting level and pending statement. Effectively, there is a small FSM inside `fhdl.dsl` implementing a parser.