m-labs/nmigen #10: Is there any reason to keep `xxx.submodules`?

mithro, 2018-12-18:
A huge number of issues in the old migen are caused by forgetting to assign something to `xxx.submodules`.

Is there any reason that we can't do something like this?
```python
class Module:
    def __setattr__(self, k, v):
        if isinstance(v, Module):
           self.add_module(k, v)
        object.__setattr__(self, k, v)
```

whitequark, 2018-12-18:
You do not inherit from Module in nMigen.

whitequark, 2018-12-18:
And incidentally the reason you don't is [this](https://github.com/m-labs/migen/blob/master/migen/fhdl/module.py#L117-L133).

whitequark, 2018-12-18:
Also, see #3--in nMigen forgetting to add a submodule will be a hard error, though I'm not yet sure exactly how this will be detected.