m-labs/nmigen #12: Implement a sanitizer for memory port combinations

whitequark, 2018-12-21:
There are a lot of possible combinations of memory ports and many of them are not legal. This depends on the vendor. In this issue we collect such behaviors to eventually implement a sanitizer, either as a part of nMigen or as a Yosys pass.

whitequark, 2018-12-21:
# Lattice iCE40
## DRAM
No distributed RAM.
## BRAM
BRAM rules:
  * Port A: read-only.
  * Port B: write-only.
  * All reads and writes are synchronous.
  * If A and B have the same address, after a simultaneous `RCLK` and `WCLK` edge, port A returns the data that was there before the write.
  * TN1256 unclear about availability of `RE`. TN1256 claims that it is only available in 256x16 configuration but lists all other primitives still with `RE` port.
  * `WE` with sub-word granularity only available in the 256x16 configuration.

nakengelhardt, 2018-12-21:
# Xilinx (7-series and ultrascale)

## DRAM

- supports asynchronous read
- if synchronous read is used, write is transparent to all other ports

- a memory description matching SP/SDP/TDP pattern is mapped to DRAM below a certain size, or if there isn't much logic in the design otherwise and the synthesizer doesn't feel like trying very hard; increasing the size of the memory may lead the synthesizer to put the same description in BRAM instead
- sometimes it will find a read register somewhere even if you intended to write an asynchronous read port

## BRAM

- the underlying BRAM hard block has two ports which can do both reads and writes.
- using the same address in the same cycle on both ports is considered an aberration. There are only two permissible cases:
  - both ports are reading, or
  - one port is writing AND it is in READ_FIRST mode, in which case the other port will read the old data. (Note that it is the setting of the *write* port that defines the behavior of the *read* port. Setting of the read port is irrelevant.)
- all other cases are user error and have undefined behavior (observing with ILA, the other port usually reads the old data).
- a port that writes will also always read simultaneously. In fact, there is no "read enable" signal, there is only a "port enable" signal and a "write enable" signal. In the xilinx examples for how to infer memories, migen's `re` is called `ena`/`enb` and always also gates writes.
- the setting of READ_FIRST/WRITE_FIRST/NO_CHANGE affects which data ends up on the read data signal of the writing port.
- there is no way to get the "fresh" data on the other port.

- when inferring any memory, unless you used the exact READ_FIRST template, vivado will assume you know that the behavior is undefined despite having written very well-defined verilog, and it can pick any behavior it wants for collisions.

whitequark, 2019-09-20:
@nakengelhardt Can you explain why does UG473 specify the BRAM dimensions for SDP mode specifically? Is it different for TDP mode?

nakengelhardt, 2019-09-20:
Do you mean this?

> Each 18 Kb block and 36 Kb block can also be configured in a simple dual-port RAM mode. In this mode, the block RAM port width doubles to 36 bits for the 18 Kb block RAM and 72 bits for the 36 Kb block RAM.

The aspect ratios supported are different for SDP and TDP mode, yes. Tables 1-11 to 1-14 on pages 29-30 have the valid combinations.