m-labs/nmigen #23: Show a better error message when "module contains unmapped RTLIL processes"

104player, 2019-01-08:
Hi, trying to learn nMigen here :)

I have run into a case where .. maybe it's as simple as I have some logic that is not used?

I get an error message like:

python oops.py generate -t v oops.v

Traceback (most recent call last):
  File "oops.py", line 34, in <module>    main(tw, ports=[])
  File "/opt/conda/lib/python3.6/site-packages/nmigen/cli.py", line 74, in main
    main_runner(parser, parser.parse_args(), *args, **kwargs)
  File "/opt/conda/lib/python3.6/site-packages/nmigen/cli.py", line 56, in main_runner    output = verilog.convert(fragment, name=name, ports=ports)
  File "/opt/conda/lib/python3.6/site-packages/nmigen/back/verilog.py", line 37, in convert
    raise YosysError(error.strip())
nmigen.back.verilog.YosysError: Warning: Module engine contains unmapped RTLIL proccesses. RTLIL processes
can't always be mapped directly to Verilog always blocks. Unintended
changes in simulation behavior are possible! Use "proc" to convert
processes to logic networks and registers.Warning: Module top contains unmapped RTLIL proccesses. RTLIL process
es
can't always be mapped directly to Verilog always blocks. Unintended
changes in simulation behavior are possible! Use "proc" to convert
processes to logic networks and registers.ERROR: Assertion failed: selection is not empty: w:* i:* %a %d o:* %a
 %ci* %d c:* %co* %a %d n:$* %d

It's not immediately apparent to a user (especially if they change more than a few lines of code at a time) which one is unused or causing this to break..

(I partially manually reduced my example code for this example)
(I have a more complex example that I think I've wired up correctly, but for some reason I need to route some signals out to pins or else it gives this same error ! (kind of like having load bearing nops) I add another signal somewhere from some verilog code, and suddenly this breaks .. sometimes even when I do wire it up to a pin -- hopefully improving the error message here will help! * Ideally let me know which signal, show why it's unmapped? )

104player, 2019-01-08:
# oops.py

from nmigen import *
from nmigen.cli import main

class ConvEngine:
    def __init__(self):
        self.coefs = Signal(8)
        self.activation = Signal(10)

    def get_fragment(self, platform):
        m = Module()
        # last element is bias
        temp = self.coefs
        m.d.comb += [
            self.activation.eq(temp),
        ]
        return m.lower(platform)

class ThingWrapper:
    def __init__(self):

        self.engine = ConvEngine()

    def get_fragment(self, platform):
        m = Module()
        m.submodules.engine = engine = self.engine

        m.d.comb += [
            engine.coefs.eq(C(50)),
        ]
        return m.lower(platform)

if __name__ == "__main__":
    tw = ThingWrapper()
    main(tw, ports=[])

104player, 2019-01-08:
[oops.zip](https://github.com/m-labs/nmigen/files/2738352/oops.zip)

whitequark, 2019-01-08:
This is actually not a user error, it's an assertion in nMigen's code that does not work correctly. It is never meant to fire on nMigen-emitted RTLIL, so if it does, it's an nMigen bug.

commit 307de722 (2019-01-08): back.verilog: remove undriven check.

This check no longer finds bugs and is prone to false positives.
Instead, we should do integration tests on the entire stack, from
fragments to Verilog.

Fixes #23.
closed 2019-01-08 by whitequark