m-labs/nmigen #24: Project Structure And Tox

peteut, 2019-01-10:
May you accept a pull request to prevent potential issues when using with `tox`?
As described in [1] best practice is to use a structure like this if used with `tox`  [2]:
```
setup.py
src/
    mypkg/
        __init__.py
        app.py
        view.py
tests/
    __init__.py
    foo/
        __init__.py
        test_view.py
    bar/
        __init__.py
        test_view.py
```
The proposed project structure is used for instance in [3]. 

[1] [Pytest:: Choosing a test layout / import rules](https://docs.pytest.org/en/latest/goodpractices.html?highlight=testpaths#choosing-a-test-layout-import-rules)
[2] [Welcome to the tox automation project](https://tox.readthedocs.io/en/latest/)
[3] [migen-axi](https://github.com/peteut/migen-axi)

whitequark, 2019-01-11:
So what is achieved here, exactly?

peteut, 2019-01-13:
@whitequark `tox` comes in handy to test against insulated Python environments where the packet to test gets installed prior testing. The configuration is stored in `tox.ini`:

```ini
[tox]
envlist = py36,py37

[base]
deps =
    -rdev-requirements.txt
    -rrequirements.txt

[testenv]
deps =
    {[base]deps}
commands =
    coverage run -m pytest {posargs}
    coverage combine .coverage
    coverage report

[travis]
python =
    3.6: py36
    3.7: py37
```

Using `tox-travis` [1] `.travis.yml` could look like this:

```yaml
sudo: false
dist: xenial
language: python
python:
  - "3.6"
  - "3.7"
cache:
  directories:
    - "$HOME/.ccache"
    - "$HOME/.local"
before_install:
  - export PATH="/usr/lib/ccache:$HOME/.local/bin:$PATH"
install:
  - pip install -rdev-requirements.txt
  - pip install coveralls
  - pip install tox-travis
  - git clone --depth=1 https://github.com/YosysHQ/yosys
  - (cd yosys && if ! yosys -V || [ $(git rev-parse HEAD $(yosys -V | awk 'match($0,/sha1 ([0-9a-f]+)/,m) { print m[1] }') | uniq | wc -l) != 1 ]; then make CONFIG=gcc ENABLE_ABC=0 PREFIX=$HOME/.local install; fi)
script:
  - tox
after_success:
  - coveralls
```

[1] [tox-travis](https://github.com/tox-dev/tox-travis)

whitequark, 2019-01-14:
This is redundant with Travis.

peteut, 2019-01-14:
> This is redundant with Travis.

That's right, Travis Python containers have only a couple of packages pre-installed which should not harm.

peteut, 2019-01-17:
Not need, as Travis is used for insulation.

closed 2019-01-17 by peteut