m-labs/nmigen #48: seperator of signal names in records causes ambiguity

anuejn, 2019-03-24:
When creating a `Record`, the signals inside it are named `recordname_fieldname` currently.

This makes it hard to see the hierarchy when using snake case names for records and fields as well. I.e. having a record named `some_record` with a field named `some_field` the resulting signal is called `some_record_some_field` which could also be the name of a signal in a nested record with depth 3. 

I propose to use another character for separation (for example `.`), which would lead to `some_record.some_field` in that case.

(see: https://github.com/m-labs/nmigen/blob/master/nmigen/hdl/rec.py#L75)

whitequark, 2019-03-25:
This ambiguity was something I expected when writing that code. The problem with `.` is that it's not a valid Verilog identifier character and so that will result in all Verilog identifiers with the dot being escaped. IMO that's about as annoying and also hard to read. But maybe we can still change that.

anuejn, 2019-03-25:
What about using the `$` sign instead? That wouldnt need to be escaped but collides with other usages of that character (however not as bad as `_` imho). Alternatively, one could use a multi character seperator like `_$` or `__`.

what do you think?

whitequark, 2019-03-25:
I think `__` makes the most sense out of these options.

anuejn, 2019-03-25:
I like that :). Should I create a PR?

whitequark, 2019-03-25:
Sure.

PR #50 (2019-03-25): Change name seperator for siganls in records to double underscore
This PR changes the seperator for signal names in records from a single underscore to a double underscore. When creating a Record, the signals inside it are then named `recordname__fieldname`.

This gives one more useful information, when looking at the signal names of fields which themselves contain underscores. E.g. `record__some_field` is now unambiguous :)

It closes #48.
closed 2019-03-25 by whitequark