m-labs/nmigen #51: strange error with arrays

rroohhh, 2019-03-26:
This 
```python
from nmigen import *
from nmigen.cli import main

m = Module()

a = Signal(1)
array = Array(x for x in range(2))

with m.If(0 == array[a]):
    m.d.sync += a.eq(0),        

if __name__ == "__main__":
    main(m, ports=[a])
```

gives a strange error (https://hastebin.com/cedapovure.sql) when running `generate`

while

```python
from nmigen import *
from nmigen.cli import main

m = Module()

a = Signal(1)
array = Array(x for x in range(2))

dummy = Signal(1)
m.d.comb += a.eq(dummy)

with m.If(0 == array[a]):
    m.d.sync += dummy.eq(0),
        

if __name__ == "__main__":
    main(m, ports=[a])
```

works

whitequark, 2019-03-26:
Yup, it's a bug and I think I know where.

rroohhh, 2019-03-26:
I don't know if it helps, but I reduced it a bit further:
```python
from nmigen import *
from nmigen.cli import main

m = Module()

a = Signal(1)
array = Array(x for x in range(2))

m.d.sync += a.eq(array[a])
        
if __name__ == "__main__":
    main(m, ports=[a])
```

fails similarly (https://hastebin.com/qujiciyihe.sql).

whitequark, 2019-03-26:
Oh, I figured it out, thanks! It's an issue with how legalizer expands arrays into switches; it does that by replacing the array index with a constant. But, in this case, you also have the array index on left-hand side, which *shouldn't* be replaced with a constant.

rroohhh, 2019-03-26:
Ah, I understand. So things like `p.part(a, 1).eq(a)` will also result in wrong verilog. 

Actually looking an the `Part` handling I think I noticed another bug: #52.

whitequark, 2019-04-21:
Thanks for the report! By the way, you can use `Array(range(2))` just as well, you don't need the `x for x in ...` part.

commit 083016d7 (2019-04-21): back.rtlil: only expand legalized values in Array/Part context on RHS.

Otherwise the following code fails to compile:

    index = Signal(1)
    array = Array(range(2))
    with m.If(0 == array[index]):
        m.d.sync += index.eq(0)

Fixes #51.
closed 2019-04-21 by whitequark