m-labs/nmigen #53: Creating signals named like verilog keywords produces invalid verilog

anuejn, 2019-03-31:
The following valid migen code...
```python
from nmigen import *
from nmigen.cli import main

input = Signal()
output = Signal()

m = Module()
m.d.comb += output.eq(input)

main(m, ports=[input, output])
```

... produces the following invalid verilog (with `python fail.py generate`:
```verilog
/* Generated by Yosys 0.8+143 (git sha1 c82aa49d, gcc 8.2.1 -march=x86-64 -mtune=generic -O2 -fno-plt -fPIC -Os) */

(* top =  1  *)
(* generator = "nMigen" *)
module top(output, input);
  (* src = "fail.py:6" *)
  reg \$next\output ;
  (* src = "fail.py:5" *)
  input input;
  (* src = "fail.py:6" *)
  output output;
  always @* begin
    \$next\output  = 1'h0;
    \$next\output  = input;
  end
  assign output = \$next\output ;
endmodule
```

I dont really know, whether this should be handled by yosys (since nMigen produces valid ir) or by nMigen itself.

programmerjake, 2019-03-31:
I would think this is a yosys bug.

whitequark, 2019-03-31:
You are using the Yosys commit *just* before I fixed that bug.

anuejn, 2019-04-01:
ah thanks for fixing :), works now :+1:

closed 2019-04-01 by anuejn