m-labs/nmigen #54: Hirarchy of submodules is not obvious from verilog

anuejn, 2019-03-31:
When adding a submodule to a submodule, the hirarchy is not obvious from the generated verilog. 

Given the first submodule is named `parent` and its child `child`, I would suggest to name the verilog submodule for the child `parent__child` instead of `child` only.

What do you think?

whitequark, 2019-03-31:
This would seriously mess up presentation in gtkwave, making it almost completely unreadable.

Maybe an attribute called `nmigen.path`? I.e. `(* nmigen.path = "parent/child" *)` as an addition to the current naming.

anuejn, 2019-04-01:
Hm... that would be an option but not that nice imho.
maybe we should consider to fix gtkwave than for that use case.

whitequark, 2019-04-01:
Sure, but gtkwave needs to be fixed first.

whitequark, 2019-04-21:
I've added an attribute. I don't think anything else can be done here without breaking gtkwave (or without gtkwave being fixed).

commit 85ae99c1 (2019-04-21): back.rtlil: emit `nmigen.hierarchy` attribute.

Fixes #54.
closed 2019-04-21 by whitequark