m-labs/nmigen #68: Tristate elaborate

osterwood, 2019-05-19:
I think there may be a typo on line 47 of Tristate.elaborate().  

https://github.com/m-labs/nmigen/blob/c337246fc581ccac0273f5661f65815266144862/nmigen/compat/fhdl/specials.py#L43-L48

The import statement at the top is:

`from .module import Module as CompatModule`

I think that Module should be CompatModule.  But, making that change and running locally results in the following error. 

```
File nmigen/compat/fhdl/module.py", line 133, in __getattr__
    .format(type(self).__name__, name))
AttributeError: 'CompatModule' object has no attribute 'd'
```

 But, that error is causing me to second guess this change.  Any ideas on what's going on here?

whitequark, 2019-05-19:
> I think that Module should be CompatModule. But, making that change and running locally results in the following error.

Why do you think that? That import is correct. (Admittedly the naming in `nmigen.compat` is confusing, but clarity was not my first priority for that part of nMigen.)

osterwood, 2019-05-19:
I meant the `m = Module()` call on line 47, not the import.  Sorry for the confusion.  

But I think I've found the actual error, an additional import statement is needed

```
from ...hdl.dsl import Module
```

line 47 is correct, but Module is not currently imported in the file.

whitequark, 2019-05-19:
Yes, that seems right.

closed 2019-05-20 by whitequark