m-labs/nmigen #79: Add a library of connector layouts, too

whitequark, 2019-06-03:
E.g. codify [this document](https://www.digilentinc.com/Pmods/Digilent-Pmod_ Interface_Specification.pdf) as `vendor.conn.pmod` or something like that.

sbourdeauducq, 2019-06-03:
Should we put this kind of vendor-specific library in nmigen or in a separate package?
(and what about the board definitions? where to draw the line?)

whitequark, 2019-06-03:
I think it should live together with the board definitions. And I think board definitions should be split. nMigen then can have proper releases, and the board/connector/etc package can be built and uploaded to PyPI each time a CI build succeeds.

whitequark, 2019-06-03:
Conversely, what we currently have as `vendor.fpga` should stay inside nMigen as the per-family primitives are fairly tied into the core of nMigen. E.g. the lowering for AsyncResetSynchronizer depends on the way attributes are processed and so on.

mithro, 2019-06-03:
FYI - I *long* time ago I started trying to figure out the various ways pmod is used. You can find a class structure I was building at https://github.com/mithro/HDMI2USB-litex-firmware-old/blob/pmod-extra/pmod.py 

I also have a spreadsheet at https://docs.google.com/spreadsheets/d/1D-GboyrP57VVpejQzEm0P1WEORo1LAIt92hk1bZGEoo/edit?usp=drive_web&ouid=100642868990821651444 which tracks a lot of details about how people use pmod.

whitequark, 2019-06-03:
@mithro But that's all just copied from the pmod spec...

mithro, 2019-06-03:
@whitequark - Yes, a lot of the data is directly from the pmod spec in a more visual / collected form that I find easier to understand. There is also a bunch of information about how people are trying to do high speed over Pmod in the later sheets ([Pmod High Speed](https://docs.google.com/spreadsheets/d/1D-GboyrP57VVpejQzEm0P1WEORo1LAIt92hk1bZGEoo/edit#gid=725118408), [Pmod High Speed Hosts](https://docs.google.com/spreadsheets/d/1D-GboyrP57VVpejQzEm0P1WEORo1LAIt92hk1bZGEoo/edit#gid=826995092), ["Pmod Like" Expansion Boards](https://docs.google.com/spreadsheets/d/1D-GboyrP57VVpejQzEm0P1WEORo1LAIt92hk1bZGEoo/edit#gid=691485833)).

whitequark, 2019-06-03:
So I imagine a library of functions like this:
```python
def PmodSPIResource(name, number, *, pmod, extras):
    return Resource(name, number,
        Subsignal("cs_n", Pins("0", dir="o", conn=("pmod", pmod))),
        Subsignal("clk",  Pins("1", dir="o", conn=("pmod", pmod))),
        Subsignal("mosi", Pins("2", dir="o", conn=("pmod", pmod))),
        Subsignal("miso", Pins("3", dir="i", conn=("pmod", pmod))),
        extras=extras
    )
```
Then it can be used:
```python
platform.add_resources([
    PmodSPIResource("spiflash", 1, pmod=0, extras={"IO_STANDARD": "SB_LVCMOS33"})
])
```
There's no need for something complicated like classes.

mithro, 2019-06-03:
@whitequark I'm not tied to the idea of classes, something like what you proposed would also work.

mithro, 2019-06-03:
I think having a standard library with these types of things in it is the important part.

commit 6426b90e (2019-06-03): vendor.conn.pmod: implement.

Fixes #79.
closed 2019-06-03 by whitequark