m-labs/nmigen #82: Global on clock pin is unroutable

zignig, 2019-06-04:
Trying to run the latest master and test the tinyFPGA_bx platform.

code 

'
from nmigen import *
from nmigen.vendor.board.tinyfpga_bx import *

class Blinky(Elaboratable):
    def elaborate(self, platform):
        clk16   = platform.request("clk16", 0)
        user_led = platform.request("user_led", 0)
        counter  = Signal(22)

        m = Module()
        m.domains.sync = ClockDomain()
        m.d.comb += ClockSignal().eq(clk16.i)
        m.d.sync += counter.eq(counter + 1)
        m.d.comb += user_led.o.eq(counter[-1])
        return m

if __name__ == "__main__":
    platform = TinyFPGABXPlatform()
    platform.build(Blinky(), do_program=True)
`
yields 

`
ERROR: BEL 'X0/Y30/io0' has no global buffer connection available
ERROR: Packing design failed.
0 warnings, 2 errors
Traceback (most recent call last):
  File "blinky.py", line 21, in <module>
    platform.build(Blinky(), do_program=True)
  File "/usr/local/lib/python3.6/dist-packages/nmigen/build/plat.py", line 108, in build
    products = plan.execute(build_dir)
  File "/usr/local/lib/python3.6/dist-packages/nmigen/build/plat.py", line 52, in execute
    subprocess.run(["sh", "{}.sh".format(self.script)], check=True)
  File "/usr/lib/python3.6/subprocess.py", line 418, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['sh', 'build_top.sh']' returned non-zero exit status 255.
`

whitequark, 2019-06-04:
That's so strange; TinyFPGA BX does not connect the clock to a global-capable pin:
[image]

closed 2019-06-04 by whitequark

zignig, 2019-06-04:
Weird , have put an issue onto the tinyfpga_bx repo.

I am also having some weirdness with clocks not propagating into submodules , will make an issue with code example when I have isolated it.

commit c89c2ce9 (2019-06-04): vendor.board.tinyfpga_bx: clk16 pin does not have a global buffer.

Fixes #82.