m-labs/nmigen #86: Platform clock definitions are very inflexible

whitequark, 2019-06-05:
Right now, platform clocks are defined like this:
```python
class TinyFPGABXPlatform(LatticeICE40Platform):
    device     = "iCE40LP8K"
    package    = "CM81"
    clocks     = [
        ("clk16", 16e6),
    ]
    resources  = [
        Resource("clk16", 0, Pins("B2", dir="i"),
                 extras={"IO_STANDARD": "SB_LVCMOS33"}),
        # ...
    ]
```
There is a problem though: it is only possible to declare a top-level resource as a clock (i.e. not a subsignal). Something like Versa has this resource in oMigen:
```python
    ("eth_clocks", 0,
        Subsignal("tx", Pins("P19")),
        Subsignal("rx", Pins("L20")),
        IOStandard("LVCMOS25")
    ),
```
which has custom code to add clock constraints in `do_finalize`.

whitequark, 2019-06-05:
I propose that instead of declaring a resource or subsignal a clock at the top level, it would be done as follows:
```python
class TinyFPGABXPlatform(LatticeICE40Platform):
    device     = "iCE40LP8K"
    package    = "CM81"
    resources  = [
        Resource("clk16", 0, Pins("B2", dir="i"),
                 extras={"IO_STANDARD": "SB_LVCMOS33"},
                 frequency=16e6),
        # ...
    ]
```
Bikeshed: maybe this would be slightly more elegant?
```python
class TinyFPGABXPlatform(LatticeICE40Platform):
    device     = "iCE40LP8K"
    package    = "CM81"
    resources  = [
        Resource("clk16", 0, Pins("B2", dir="i"), Clock(16e6),
                 Extras(IO_STANDARD="SB_LVCMOS33")),
        # ...
    ]
```

cr1901, 2019-06-05:
Just to add a data point: Some platforms, TinyFPGA A in particular, don't actually _have_ a clock pin, instead relying on the FPGA's internal oscillator. I did some [things](https://github.com/m-labs/migen/blob/master/migen/build/platforms/tinyfpga_a.py#L107-L118) I'm not exactly proud of to get it to play nice with `omigen`, even if the commit was accepted. 

So if there is a way to more nicely support internal clock constraints or otherwise without a `do_finalize` dance or `request` overload, I'm open to feedback.

whitequark, 2019-06-05:
Right, so that's a related but different situation: defining frequency constraints on ports and on nets. Currently nextpnr is deficient in that it does not allow to reliably define a frequency constraint on a net, so we're going to have to teach `platform.add_frequency_constraint` to distinguish ports and nets, and only support for ports will be an absolute requirement for any new platform.

commit c9879c79 (2019-06-05): build.{dsl,res,plat}: apply clock constraints to signals, not resources.

This adds the Clock() build DSL element, and adds a resource manager
function add_clock_constraint() that takes a Pin or a Signal.
Note that not all platforms, in particular not any nextpnr platforms
at the moment, can add constraints on arbitrary signals.

Fixes #86.
closed 2019-06-05 by whitequark