m-labs/nmigen #88: Signal names are not translated for constraints, etc.

whitequark, 2019-06-05:
This is technically a bug in all cases, but it is not actually visible in most circumstances, as the signals that appear in constraint files are always ports, ports are assigned names first, and port names are never duplicates when returned by the platform layer. However, it would break most attempts to add interior frequency constraints, like for PLL output.

A complication is that it is currently not possible to reliably add constraints on interior signals in nextpnr (and even ports became reliable only after #71).

peteut, 2019-06-15:
For *Vivado* non-related clock domains have to explicitly marked as such (for instance using `set_clock_groups .. -asynchronous`). To accomplish that I'm naively using `<clock_pad>.name` to compose the constraint. Is it okay to assume pad names are resolved to `<pad>.name`?

whitequark, 2019-09-11:
Since 8c30147e3, constraints on interior signals are supported if the toolchain supports them.

whitequark, 2019-09-14:
This should be fixed in https://github.com/YosysHQ/nextpnr/pull/329, but needs some adjustments to `vendor` files.

whitequark, 2019-09-20:
cc @dlharmon

commit 91ef2f58 (2019-09-20): vendor.lattice_{ecp5,ice40}: allow clock constraints on arbitrary signals.

Fixes #88.
closed 2019-09-20 by whitequark