m-labs/nmigen #92: Using Yosys with Xilinx Series 7 devices (and non-standard tool flows)

mithro, 2019-06-06:
How do you suggest adding support for using Yosys for Synthesis and Vivado for place and route to the https://github.com/m-labs/nmigen/blob/master/nmigen/vendor/xilinx_7series.py file?

I'm wondering if it makes sense to split the build stuff into separate synthesis and place and route tooling?

The goal is that Yosys should be able to accept the exact same Verilog as Vivado (and the other commercial tools), so in **theory** nmigen doesn't need to change the Verilog output?

Alternatively, if Yosys is the primary frontend and nmigen is already using Yosys directly, is there something better that could be done?

I'm interested in;
 - [ ] (Series 7) Yosys (synthesis) -> Vivado (place and route + bitgen)
 - [ ] (Series 7) Yosys (synthesis) -> VtR (place and route + bitgen)
 - [ ] (Series 7) Yosys (synthesis) -> nextpnr (place) -> RapidWright + Vivado (route + bitgen)
 - [ ] (Spartan 6) Yosys (synthesis) -> ISE (place and route + bitgen)
 - [ ] (ECP5) Yosys (synthesis) -> nextpnr (place and route + bitgen)
 - [ ] (ECP5) Yosys (synthesis) -> Diamond (place and route + bitgen)
 - [ ] (Igloo2) Yosys (synthesis) -> Libero (place and route + bitgen)

See diagram at https://j.mp/openfpga-diagram and below;
[image]

whitequark, 2019-06-07:
> How do you suggest adding support for using Yosys for Synthesis and Vivado for place and route to the https://github.com/m-labs/nmigen/blob/master/nmigen/vendor/xilinx_7series.py file?

I don't. I think what you're suggesting will result in a compatibility nightmare for essentially no practical benefit. In my opinion, supporting 4 distinct flows, each with their own set of obscure bugs, for Series 7 alone (much less doing this for every conceivable target) is an anti-goal.

whitequark, 2019-06-07:
> * [ ]  (ECP5) Yosys (synthesis) -> nextpnr (place and route + bitgen)

Note that this is an exception in this list, since it's an officially supported and reasonably stable already flow. I actually have most code for it already, I just need to polish it.

mithro, 2019-06-07:
@whitequark 

Do you have suggestions for the most maintainable way to extend the `nmigen.vendor` flow as an external party? Does it make sense to just subclass the existing upstream (this repositories) vendor flows? 

The code already has a lot of flexibility thanks to the `script_(before|after)_xxx` hooks but I can't see an easy way to just override the synthesis part?

I would like to have a good set of IP and applications which can be used to compare the open flows to the vendor versions. As MiSoC + LiteX are some of the most advanced users who truly care about open source I would like to see the open flows working really well with them.

whitequark, 2019-06-07:
@mithro 

I think it'd be perfectly fine to override both file and command templates. The Yosys synthesis part is very minimal and I deliberately duplicate it among in-tree platforms because that makes it easier to understand how exactly it works. You can just copy and paste it as it's not really changing much if at all.

whitequark, 2019-06-25:
ECP5 with Yosys and nextpnr is done.

whitequark, 2019-06-28:
I thought more about this issue, and came to two conclusions:

  1. Supporting a combinatorial product of toolchain components is not desirable *per se*: it has many drawbacks described above, and in general I would expect that adding one more supported configuration would make the user experience *worse* by ensuring there will be *more bugs*.
  2. Nevertheless, supporting *experimental* combinations of toolchain components is a useful goal for reasons also described above. Indeed, that is how a new flow will come into existence in the first place.

Now, given these conclusions, I do not want to make any changes to existing flows solely to support these many new combinations that will almost never be used in practice. However, I am open to making changes to existing flows that are *generally useful* (for example, factoring out IOB instantiation code). To understand exactly which changes should be made, I would need to examine the particular flows in detail, and consider this in the wider context of nMigen design issues.

If you have a specific flow in mind that you would like to support, I encourage you to add this flow to nMigen and report your experience and pain points so that the nMigen design can be adjusted to accomodate this. (Of course, you can also contract me to do so.) However, I will not be adding any features to support non-standard tool flows that are not justified by real-world experience (i.e. working code that is not as good as it should be).

closed 2019-06-28 by whitequark

sbourdeauducq, 2019-06-28:
Replacing the Vivado synthesis front-end with Yosys may be useful (or may not, if there are other problems with Yosys) to work around certain [bugs](https://github.com/m-labs/artiq/issues/1230) that Xilinx are taking forever to fix.

whitequark, 2019-06-28:
As far as I know, at this point Yosys still cannot infer many primitives that Vivado can, so it is unlikely that ARTIQ will pass timing. That said it is an interesting experiment, though of course ARTIQ will need to be ported to nMigen first (which has many dependencies like a MiSoC port.)

mithro, 2019-06-28:
FYI, @eddiehung is working hard to make a LiteX SoC with VexRISC-V, DDR and Ethernet on Artix 7 work with a Yosys->Vivado flow.