m-labs/nmigen #94: Platform builds do not add submodule devices

zignig, 2019-06-09:
Attempting to add a larger core as a submodule for a platform build and the build process seems to optimise the cores out for some reason. 

```
zignig@noid:/opt/FPGA/tinybx_stuff/vanishing$ python minerv.py
Warning: No clocks found in design
Warning: No clocks found in design
```

Source and build artefacts are in https://github.com/zignig/tinybx_stuff/tree/master/vanishing

From my limited knowledge of the internals, I have noticed a a few things. From the build report  (top.rpt)

hierarchy looks good 
```
2.2.1. Analyzing design hierarchy..
Top module:  \top
Used module:     \minerva
Used module:         \shifter
Used module:         \predict
Used module:         \logic
Used module:         \loadstore
Used module:             \anonymous$1
Used module:         \fetch
Used module:             \anonymous
Used module:         \exception
Used module:             \interrupt_pe
Used module:             \trap_pe
Used module:         \decoder
Used module:         \compare
Used module:         \adder
Used module:         \csrf
Used module:         \w
Used module:         \m
Used module:         \x
Used module:         \d
Used module:         \f
Used module:         \a

```
snip 

the in the PROC_CLEAN pass it seems to remove all the processes 

```
Removing empty process `top.$group_0'.
Found and cleaned up 2 empty switches in `\minerva.$group_253'.
Removing empty process `minerva.$group_253'.
Found and cleaned up 2 empty switches in `\minerva.$group_252'.
```
snip

and then removes all the submodules 

```
No more expansions possible.
Deleting now unused module minerva.
Deleting now unused module shifter.
Deleting now unused module predict.
Deleting now unused module logic.
Deleting now unused module loadstore.
Deleting now unused module anonymous$1.
Deleting now unused module fetch.
Deleting now unused module anonymous.
Deleting now unused module exception.
Deleting now unused module interrupt_pe.
Deleting now unused module trap_pe.
Deleting now unused module decoder.
Deleting now unused module compare.
Deleting now unused module adder.
Deleting now unused module csrf.
Deleting now unused module w.
Deleting now unused module m.
Deleting now unused module x.
Deleting now unused module d.
Deleting now unused module f.
Deleting now unused module a.
```
Then leaves an empty design

```
=== top ===

   Number of wires:               1102
   Number of wire bits:           7272
   Number of public wires:        1102
   Number of public wire bits:    7272
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:                  1
     SB_IO                           1
```

nextpnr , yosys , nmigen , nmigen-boards , minerva are all latest master.

This effect is repeated with the boneless example ( using my Boneless-CPU branch )

whitequark, 2019-06-09:
What's happening here is that your design only has inputs, not outputs. Since none of it does anything that would have an observable effect, it makes no difference if the entire design is removed. Which is what Yosys correctly does.

zignig, 2019-06-09:
Added a LED output to the boneless, and shazam , it works. Will remember for later. Thanks

closed 2019-06-09 by zignig