m-labs/nmigen #97: Bikeshed: conventions for CDC primitives

whitequark, 2019-06-11:
One reason I've hesitated to merge #40 so far is that I think the oMigen conventions for CDC primitives are a bit annoying. I think a better situation than the current one would be:
  * Instead of arguments named `domain`, `idomain`, `rdomain`, etc, which only allow a single letter for domain designation, use `cd`, `cd_i`, `cd_read`.
  * All modules which are likely to be transformed with `DomainRenamer` right after creation should allow changing the domain names through arguments. Notably this would include `AsyncFIFO`, which is probably responsible for most uses of `DomainRenamer`, but all other CDC primitives as well. Non-CDC modules may or may not allow that depending on intended use.

Thoughts? @sbourdeauducq @jordens @Wren6991

sbourdeauducq, 2019-06-11:
Sounds ok.

Wren6991, 2019-06-28:
I like this, but find some of the results inconsistent (c84cd8d):

```
class Gearbox(Elaboratable):
   ...
    def __init__(self, iwidth, cd_i, owidth, cd_o):
```

Inconsistent use of suffix vs prefix is something I would trip over when typing too fast.

whitequark, 2019-06-28:
I think `Gearbox` should of course use `width_i` and `width_o` here.

commit bdb70ad4 (2019-09-13): lib.fifo: adjust for new CDC primitive conventions.

Fixes #97.
closed 2019-09-13 by whitequark