m-labs/nmigen #98: Generated Verilog should be more readable

whitequark, 2019-06-11:
This is solely blocked on Yosys issue yosyshq/yosys#726. It's in my queue for some time, but the threshold for merging it is fairly high (multiple days of randomized testing), so I haven't been able to push it to completion yet.

mithro, 2019-06-11:
@whitequark Would having access to a bunch of CPU resources help with the randomized testing?

whitequark, 2019-06-11:
@mithro It's kind of a pain to set up VlogHammer in the first place. I think I can use the M-Labs machine for testing that once I have a general confidence in the correctness of the changes to that pass.

whitequark, 2019-07-14:
> Would having access to a bunch of CPU resources help with the randomized testing?

Actually, what kind of resources can you provide?

mithro, 2019-07-15:
I could provide pretty much anything on the list at https://cloud.google.com/compute/docs/machine-types

Anything like the following might work?
 * n1-standard-96
 * n1-highmem-64
 * n1-highcpu-96

whitequark, 2019-07-31:
Fixing YosysHQ/yosys#726 proved to be extraordinarily complex, so bumping this from 0.1 milestone.