m-labs/nmigen #100: nmigen.compat Case.makedefault() broken

sbourdeauducq, 2019-06-12:
Minimal repro:
```python
from nmigen.compat import *

mux = Signal(4)
cases = {0: []}
Case(mux, cases).makedefault()
```

Result:
```
Traceback (most recent call last):
  File "foo.py", line 5, in <module>
    Case(mux, cases).makedefault()
  File "/nix/store/vqlpzfgw1k2k5pgby8g7b1kgc4b13g7w-python3-3.7.3-env/lib/python3.7/site-packages/nmigen/tools.py", line 73, in wrapper
    return f(*args, **kwargs)
  File "/nix/store/vqlpzfgw1k2k5pgby8g7b1kgc4b13g7w-python3-3.7.3-env/lib/python3.7/site-packages/nmigen/compat/fhdl/structure.py", line 107, in makedefault
    key = "{:0{}b}".format(wrap(key).value, len(self.test))
  File "/nix/store/vqlpzfgw1k2k5pgby8g7b1kgc4b13g7w-python3-3.7.3-env/lib/python3.7/site-packages/nmigen/tools.py", line 73, in wrapper
    return f(*args, **kwargs)
  File "/nix/store/vqlpzfgw1k2k5pgby8g7b1kgc4b13g7w-python3-3.7.3-env/lib/python3.7/site-packages/nmigen/compat/fhdl/structure.py", line 18, in wrap
    return Value.wrap(v)
  File "/nix/store/vqlpzfgw1k2k5pgby8g7b1kgc4b13g7w-python3-3.7.3-env/lib/python3.7/site-packages/nmigen/hdl/ast.py", line 41, in wrap
    raise TypeError("Object '{!r}' is not an nMigen value".format(obj))
TypeError: Object ''0000'' is not an nMigen value
```

sbourdeauducq, 2019-06-12:
The issue comes from ``Case`` turning case values into strings when they are added.

whitequark, 2019-06-12:
Hm, I think a lot of that logic is fairly broken, like non-last "default".

whitequark, 2019-06-13:
Okay, fixed that and everything else I could see, I think.

commit 25667470 (2019-06-13): compat.fhdl.structure: fix Case().makedefault().

Fixes #100.
closed 2019-06-13 by whitequark