m-labs/nmigen #107: Using Verilog attribute on Instance

DurandA, 2019-06-24:
I'd like to use Verilog’s `(* attribute *)` annotations on `Instances` (to add [constraints](https://github.com/YosysHQ/nextpnr/blob/master/docs/constraints.md)).

As this feature is not implemented yet, I will try to add it myself:
* Would you accept a PR with such a feature?
* Since I am not familiar with the FHDL->Verilog generation process, could you provide me some rough guidelines on how to implement this?

whitequark, 2019-06-24:
I think I have the code for that somewhere, it just needs some tests.

whitequark, 2019-06-24:
<details>
<summary>Instance attributes</summary>

```python
diff --git a/nmigen/back/rtlil.py b/nmigen/back/rtlil.py
index 1a1fabc..9bd7da9 100644
--- a/nmigen/back/rtlil.py
+++ b/nmigen/back/rtlil.py
@@ -106,9 +106,11 @@ class _ModuleBuilder(_Namer, _Bufferer):
         self._append("  memory width {} size {} {}\n", width, size, name)
         return name
 
-    def cell(self, kind, name=None, params={}, ports={}, src=""):
+    def cell(self, kind, name=None, params={}, ports={}, attrs={}, src=""):
         self._src(src)
         name = self._make_name(name, local=False)
+        for attr_name, attr_value in attrs.items():
+            self.attribute(attr_name, attr_value)
         self._append("  cell {} {}\n", kind, name)
         for param, value in params.items():
             if isinstance(value, str):
@@ -678,7 +680,7 @@ def convert_fragment(builder, fragment, hierarchy):
             return "\\{}".format(fragment.type), port_map
 
     module_name  = hierarchy[-1] or "anonymous"
-    module_attrs = {}
+    module_attrs = OrderedDict()
     if len(hierarchy) == 1:
         module_attrs["top"] = 1
     module_attrs["nmigen.hierarchy"] = ".".join(name or "anonymous" for name in hierarchy)
@@ -760,7 +762,8 @@ def convert_fragment(builder, fragment, hierarchy):
                     compiler_state.resolve_curr(signal, prefix=sub_name)
                 sub_ports[port] = rhs_compiler(value)
 
-            module.cell(sub_type, name=sub_name, ports=sub_ports, params=sub_params)
+            module.cell(sub_type, name=sub_name, ports=sub_ports, params=sub_params,
+                        attrs=subfragment.attrs)
 
         # If we emit all of our combinatorial logic into a single RTLIL process, Verilog
         # simulators will break horribly, because Yosys write_verilog transforms RTLIL processes
diff --git a/nmigen/hdl/ir.py b/nmigen/hdl/ir.py
index a49db77..d2dcbdc 100644
--- a/nmigen/hdl/ir.py
+++ b/nmigen/hdl/ir.py
@@ -71,6 +71,7 @@ class Fragment:
         self.statements = []
         self.domains = OrderedDict()
         self.subfragments = []
+        self.attrs = OrderedDict()
         self.generated = OrderedDict()
         self.flatten = False
 
diff --git a/nmigen/hdl/xfrm.py b/nmigen/hdl/xfrm.py
index 7b66764..e73c18e 100644
--- a/nmigen/hdl/xfrm.py
+++ b/nmigen/hdl/xfrm.py
@@ -271,6 +271,7 @@ class FragmentTransformer:
         else:
             new_fragment = Fragment()
             new_fragment.flatten = fragment.flatten
+        new_fragment.attrs = OrderedDict(fragment.attrs)
         self.map_ports(fragment, new_fragment)
         self.map_subfragments(fragment, new_fragment)
         self.map_domains(fragment, new_fragment)
```
</details>

DurandA, 2019-06-26:
Your patch works great, thanks!

Should I close this issue or do you plan to merge this feature?

whitequark, 2019-06-26:
I'll merge it soon.

commit 48d4ee40 (2019-06-28): hdl.ir, back.rtlil: allow specifying attributes on instances.

Fixes #107.
closed 2019-06-28 by whitequark