m-labs/nmigen #113: Towards release 0.1

whitequark, 2019-06-28:
With the core language feature-complete, the build system nearing maturity, and the library gaining many important primitives, it is time to indicate that nMigen can be used in practical designs by people not closely involved in its development. (I've heard that there is a lot of interest in that.) In this issue I outline what I see as prerequisites to [milestone 0.1](https://github.com/m-labs/nmigen/milestone/1).

  1. We need a Yosys 0.9 release, since Yosys 0.8 contains a number of blocker bugs like ~#53~. We should check for Yosys version in `back.verilog` as well, which is #55.
  2. Creation of clock domains (or lack thereof) is an important UI issue. In particular, it seems like everyone gets domain propagation wrong at first, so if people don't have to explicitly create `ClockDomain` objects *either* to convert an isolated module to Verilog *or* to use the platform with just a single clock domain, it would eliminate a lot of beginner papercuts. That's #57. 
  3. Although not an issue that affects semantics, generated Verilog being unreadable is a significant annoyance to people who has an easier time reading Verilog than RTLIL, i.e. basically everyone. That's #98. Unfortunately it missed the merge window for Yosys 0.9, so we'll have to highlight it in README, emit a warning (ugh), or something like that.
  4. Regardless of the standard library items that should be *added* to reach 0.1 (which is subject to some discussion), the standard library items that are *already* in master should not have their most basic interface change after 0.1. That means applying the new convention from #97.
  5. @Wren6991 did some very good work porting CDC primitives from oMigen, and it would be a shame if it missed 0.1, especially as there are no outstanding issues I can see, and it would give us a big improvement towards oMigen compatibility.
  6. COMPAT_SUMMARY will need to be updated; this is #112.
  7. Simulator bugs (#28, #34, #47) should at least be triaged, since right now it is not even clear why they are happening.

Now there are some issues that I think shouldn't block 0.1, but I'm open to feedback:
  1. Synchronous transparent read ports (#16); lack of such ports was a problem in Minerva, so it does come up in real designs. Unfortunately this both misses the Yosys 0.9 merge window, and is quite nontrivial to actually implement. I think it's more of a 0.2 issue.
  2. CDC improvements like CDC safety (#4) and automatic false path constraints (#87; depends on #88). These are definitely needed, but there are still plenty of designs that will be just fine without them, and I think for most people waiting to pick up nMigen they wouldn't be an issue. Again, these are more of 0.2 issues.

Every other outstanding issue is more of a nice-to-have and shouldn't count here.

whitequark, 2019-06-28:
@sbourdeauducq @jordens Feedback?

sbourdeauducq, 2019-07-10:
Sounds good to me.

>     3\. Although not an issue that affects semantics, generated Verilog being unreadable is a significant annoyance to people who has an easier time reading Verilog than RTLIL, i.e. basically everyone.

There are also people who will judge the tool based on the quality of the generated Verilog.

whitequark, 2019-07-10:
That is indeed true. And I regularly improve Yosys to do better, like in 10e56c75.

whitequark, 2019-09-18:
Most of the issues here have been resolved. Since I will likely rewrite the simulator for 0.2, simulator bugs (#28, #34, #47) are postponed until that point. The remaining [issues](https://github.com/m-labs/nmigen/milestone/1) in the 0.1 milestone should be the last ones that I expect to add.

closed 2019-09-18 by whitequark