m-labs/nmigen #116: Should memory read ports use domain="comb" or domain=None instead of synchronous=False?

whitequark, 2019-06-29:
It feels odd to have the `memory.read_port(domain="sync", synchronous=False)` combination of arguments to be valid, and in fact that seems like it'd cause some hard to find bugs.

sbourdeauducq, 2019-07-03:
Why not ``domain=None``? ``comb`` is a valid, albeit ill-chosen, name for a synchronous domain.

whitequark, 2019-07-03:
@sbourdeauducq Because of the way `Module` is implemented, trying to do `m.d.comb += stmt` or `m.d["comb"] += stmt` will always add those statements to the combinatorial domain, even if there is a synchronous domain called `comb`.

So I wouldn't say it's valid, and further, we probably need to disallow either creating a `ClockDomain("comb")` or adding it to any module.

whitequark, 2019-07-03:
Closing per IRC discussion.

commit 94e8f479 (2019-07-01): hdl.mem: use read_port(domain="comb") for asynchronous read ports.

This avoids the absurdity of the combination of arguments that is
read_port(domain="sync", synchronous=True).

Fixes #116.
closed 2019-07-03 by whitequark