m-labs/nmigen #118: Unable to create a ring oscillator

DurandA, 2019-07-01:
I am trying to build a ring oscillator (targeting iCE40) using nmigen. I first made a simple Verilog design (see above, adapted from [Clifford's ringosc.v](http://svn.clifford.at/handicraft/2015/ringosc/ringosc.v)) and then tried to port it to nmigen:

```python
class RingOscillator(Elaboratable):
    def elaborate(self, platform):
        user_led = platform.request("user_led", 0)

        m = Module()

        cd_chain = ClockDomain(async_reset=True)
        m.domains += cd_chain

        buffers_in = Signal(7)
        buffers_out = Signal(7)
        
        m.d.comb += buffers_in.eq(buffers_out[-1:] + buffers_out[0:-1])
        
        for buf_in, buf_out in zip(buffers_in, buffers_out):
            inverter = Instance("SB_LUT4",
                                    p_LUT_INIT=0b01,
                                    i_I0=buf_in,
                                    i_I1=Const(0),
                                    i_I2=Const(0),
                                    i_I3=Const(0),
                                    o_O=buf_out)
            m.submodules += inverter

        m.d.comb += cd_chain.clk.eq(buffers_out[-1])
        
        counter  = Signal(24)
        m.d.chain += counter.eq(counter + 1)
        m.d.comb += user_led.o.eq(counter[-1])

        return m
```

Unfortunately, the above code doesn't produce a working oscillator. It is unclear to me if I don't understand how to use nmigen clocks or if this is due to unwanted optimizations (or if I made a dumb mistake somewhere else).

Here is a simple working version in Verilog:

```verilog
module top (
    output LED
);
    wire [6:0] buffers_in, buffers_out;
    assign buffers_in = {buffers_out[5:0], buffers_out[6]};

    SB_LUT4 #(
        .LUT_INIT(16'd1)
    ) buffers [6:0] (
        .O(buffers_out),
        .I0(buffers_in),
        .I1(1'b0),
        .I2(1'b0),
        .I3(1'b0)
    );

    reg [23:0] counter = 0;
    always @(posedge buffers_out[6]) begin
        counter <= counter + 1;
    end

    assign LED = counter[23];

endmodule
```

For both versions, nextpnr should be configured with `--ignore-loops` switch.

How to use a proper clock attached to the oscillator? Is this problem related to #57?

peteut, 2019-07-01:
@DurandA did you forgot to assign `ResetSignal`, either using `ResetSignal("chain")` or `cd_chain.rst`?

peteut, 2019-07-01:
```python
m.d.comb += buffers_in.eq(buffers_out[-1:] + buffers_out[0:-1])
```
should spell
```python
m.d.comb += buffers_in.eq(Cat(buffer_out[-1], buffers_out[:-1]))
```
to reflect the Verilog code.

DurandA, 2019-07-01:
> ```python
> m.d.comb += buffers_in.eq(buffers_out[-1:] + buffers_out[0:-1])
> ```
> 
> should spell
> 
> ```python
> m.d.comb += buffers_in.eq(Cat(buffer_out[-1], buffers_out[:-1]))
> ```
> 
> to reflect the Verilog code
> 
> ```verilog
> assign buffers_in = {buffers_out[5:0], buffers_out[6]};
> ```

Many thanks, this solved the issue. I thought that Python slices supported concatenation.

closed 2019-07-01 by DurandA

whitequark, 2019-07-01:
> I thought that Python slices supported concatenation.

Python doesn't have a slice *object*\*, it has a slice *operator*. The slice operator applied to lists returns a list, and list's `+` operator performs concatenation. The slice operator applied to an nMigen value returns an nMigen value, and nMigen value's `+` operator performs addition.

<sub>\* There is the `slice` class, but it's more of an implementation detail of the `__getitem__` function than an actual language feature, and is not relevant here.</sub>

DurandA, 2019-07-01:
> Python doesn't have a slice _object_*, it has a slice _operator_. The slice operator applied to lists returns a list, and list's `+` operator performs concatenation. The slice operator applied to an nMigen value returns an nMigen value, and nMigen value's `+` operator performs addition.

Thank you for the clarification! This is very much appreciated.