m-labs/nmigen #124: nextpnr error: Cell cannot be bound to bel since it is already bound

sbourdeauducq, 2019-07-03:
Repro:
```python
from nmigen import *
from nmigen_boards.versa_ecp5 import VersaECP5Platform

class Top(Elaboratable):
    def elaborate(self, platform):
        m = Module()

        foo = Signal()
        m.d.comb += platform.request("uart").tx.o.eq(0)
        
        return m

if __name__ == "__main__":
    VersaECP5Platform().build(Top())
```

```
$ python crash.py 
ERROR: Cell '$2' cannot be bound to bel 'X38/Y0/PIOB' since it is already bound to cell 'uart_0__rx__io$tr_io'
```

This is with nextpnr ff958830d1097b9bfa3c3b34094e671741ef563d, Yosys 8455d1f4ffb942c802b65e20748e54a123e08df0 and nMigen 9a1048af50aa2f7b8b75260832e8849255190253.

whitequark, 2019-07-03:
That was caused by a typo (m-labs/nmigen-boards@6e0cc9d) that caused `rx` and `tx` to be bound to the same ball. The error message would be much better with #123, and also we should probably detect this before handing over the RTL to the vendor toolchain.

whitequark, 2019-07-03:
@sbourdeauducq With #123, the message now looks like:
```
$ python3 crash.py
ERROR: Cell 'uart_0__tx' cannot be bound to bel 'X38/Y0/PIOB' since it is already bound to cell 'uart_0__rx__io$tr_io'
```

whitequark, 2019-07-03:
With latest master, the message now looks like:
```
$ python3 crash.py
Traceback (most recent call last):
  File "t.py", line 16, in <module>
    VersaECP5Platform().build(Top())
  File "/home/whitequark/Projects/nmigen/nmigen/build/plat.py", line 47, in build
    plan = self.prepare(fragment, name, **kwargs)
  File "/home/whitequark/Projects/nmigen/nmigen/build/plat.py", line 61, in prepare
    fragment = Fragment.get(fragment, self)
  File "/home/whitequark/Projects/nmigen/nmigen/hdl/ir.py", line 56, in get
    obj = obj.elaborate(platform)
  File "t.py", line 10, in elaborate
    m.d.comb += platform.request("uart").tx.o.eq(0)
  File "/home/whitequark/Projects/nmigen/nmigen/build/res.py", line 151, in request
    attrs=resource.attrs)
  File "/home/whitequark/Projects/nmigen/nmigen/build/res.py", line 111, in resolve
    attrs={**attrs, **sub.attrs})
  File "/home/whitequark/Projects/nmigen/nmigen/build/res.py", line 135, in resolve
    .format(name, phys_name, self._phys_reqd[phys_name]))
nmigen.build.res.ResourceError: Resource component uart_0__tx uses physical pin C11, but it is already used by resource component uart_0__rx that was requested earlier
```

commit b471e33d (2019-07-03): build.res: detect physical conflicts earlier.

This is useful for two reasons:
  1. nMigen can provide better error messages than the platform and
     do it earlier in the build pipeline.
  2. Many platforms handle diffpairs by only constraining the P pin;
     the N pin is completely ignored. If this is undetected,
     downstream users (human or software) can rely on this
     information assuming it is correct and introduce more errors.
     (Of course, this will not catch every mistake, but the most
     common is a copy-paste issue, and that will handle it.)

Fixes #124.
closed 2019-07-03 by whitequark