m-labs/nmigen #125: "comb" should not be accepted as a name in ClockDomain(), DomainRenamer(), etc

whitequark, 2019-07-03:
See #116 for motivation.

sbourdeauducq, 2019-07-03:
Why not just use ``None`` instead in memory read ports?

whitequark, 2019-07-03:
@sbourdeauducq But that would still be a problem. Consider this code:
```
m = Module()
m.d.comb += a.eq(1)
m2 = DomainRenamer({"comb":"sync"})(m)
```
It wouldn't work the same as:
```
m2 = Module()
m2.d.sync += a.eq(1)
```
In fact this use of `DomainRenamer` is currently a no-op.

So regardless of the decision for #116, *something* needs to be fixed there, one way or another. (You could argue that this use of `DomainRenamer` should be valid--although I find it unconvincing--and then it would need to be fixed to accept such code.)

commit a7fbff94 (2019-07-08): hdl.{ast,cd,dsl,xfrm}: reject inappropriately used comb domain.

Fixes #125.
closed 2019-07-08 by whitequark