m-labs/nmigen #131: `TOOLCHAIN_ENV` user variable

cr1901, 2019-07-07:
One convenient thing `omigen` had that was [dropped](https://github.com/m-labs/migen/commit/dd4ed5d0c5a234caa9ba690171d2593d40c407db) recently was the ability to inspect the user's system for toolchains not normally on the path and create a `bat`/`sh` file to automatically load the correct environment for these toolchains.

We have discussed bringing this feature back in a more limited form via an environment variable that a user can set called `TOOLCHAIN_ENV`.

## How It Works
* On Linux, if  `TOOLCHAIN_ENV` is set and not empty (including empty string), the Jinja2 renderer will prepend the generated `sh` file with a line that `source`s the file that  `TOOLCHAIN_ENV` points to.
* On Windows, if  `TOOLCHAIN_ENV` is set and not an empty string, the Jinja2 renderer will prepend the generated `bat` file with a line that `call`s the file that  `TOOLCHAIN_ENV` points to.

## Open Questions
* Should this var be tailored for each toolchain nmigen supports?
* Should this var be split into `TOOLCHAIN_SOURCE` and `TOOLCHAIN_CALL`, since we unconditionally generate `sh` and `bat` files regardless of host system?

whitequark, 2019-07-07:
> * On Linux, if  `TOOLCHAIN_ENV` is set and not empty (including empty string), the Jinja2 renderer will prepend the generated `sh` file with a line that `source`s the file that  `TOOLCHAIN_ENV` [...]

I think this isn't quite right. Build plans can be archived and transferred, say if you need to show an MCVE, and in that case it is the *recipient* who should be able to configure the build plan. (To a lesser extent this might apply to `NMIGEN_verbose` and `NMIGEN_quiet`, arguably.) So it should really be checked by the script itself.

cr1901, 2019-07-07:
> I think this isn't quite right. Build plans can be archived and transferred, say if you need to show an MCVE, and in that case it is the recipient who should be able to configure the build plan. (To [...]
Good point.

>Should this var be split into TOOLCHAIN_SOURCE and TOOLCHAIN_CALL, since we unconditionally generate sh and bat files regardless of host system?
In that case `TOOLCHAIN_{vivado, diamond, trellis, ise, etc}_ENV` seems very reasonable.

commit 1fcd495b (2019-07-07): build.plat: source a script with toolchain environment.

Fixes #131.
commit 146f3cb6 (2019-07-07): build.plat: source a script with toolchain environment.

Fixes #131.
closed 2019-07-07 by whitequark