m-labs/nmigen #132: Allow dependent platform attributes

whitequark, 2019-07-07:
It's fairly common that developer boards have banks with switchable voltage, e.g. https://github.com/m-labs/nmigen-boards/pull/15. It is definitely possible to add resources in `__init__` to set these up, but it is more elegant to support this natively, using something like:
```python
resources = [
    Resource("user_led", 7, ..., Attrs(IOSTANDARD=bank2_iostandard)),
]

def bank2_iostandard(self):
    if self.JP12 == "2V5":
        return "LVCMOS25"
    else:
        return "LVCMOS33"
```

mithro, 2019-07-07:
FYI - I did something very hacky like this here -> https://github.com/timvideos/litex-buildenv/blob/de58bffaa909696827ca552f392f034c57c09467/platforms/atlys.py#L8-L45 to also support the Digilent Atlys board.

whitequark, 2019-07-07:
That's pretty much the same thing as I suggest but more verbose.

commit 7b4fbf8e (2019-07-08): build.{dsl,res}: allow platform-dependent attributes using callables.

Fixes #132.
closed 2019-07-08 by whitequark