m-labs/nmigen #134: Add FSM state name annotations to generated Verilog output

mithro, 2019-07-07:
Currently when using an FSM, the generate Verilog looks something like;
```verilog
  always @* begin
    \current_word$next  = current_word;
    casez (fsm_state)
      2'h0:            
          /* empty */;
      2'h1:                  
        begin          
          \current_word$next [0] = io_miso[0];
          \current_word$next [1] = io_miso[1];
          \current_word$next [2] = io_miso[2];
          \current_word$next [3] = io_miso[3];
          \current_word$next [4] = io_miso[4];
          \current_word$next [5] = io_miso[5];
          \current_word$next [6] = io_miso[6];
          \current_word$next [7] = io_miso[7];
        end                  
      2'h2:            
        begin                     
          \current_word$next [4] = io_miso[0];
          \current_word$next [5] = io_miso[1];
          \current_word$next [6] = io_miso[2];
          \current_word$next [7] = io_miso[3];
        end
      2'h3:            
        begin    
          \current_word$next [6] = io_miso[0];
          \current_word$next [7] = io_miso[1];
        end                       
    endcase               
    casez (rst)                               
      1'h1:            
          \current_word$next  = 8'h00;
    endcase
  end   
```

If would be nice if the states in the case statement had some type of comment or other annotation which signified their names.

Using something like comments;
```verilog
  always @* begin
    \current_word$next  = current_word;
    casez (fsm_state)
      2'h0:            /* IDLE */
          /* empty */;
      2'h1:            /* SPI_x8 */      
        begin          
          \current_word$next [0] = io_miso[0];
          \current_word$next [1] = io_miso[1];
          \current_word$next [2] = io_miso[2];
          \current_word$next [3] = io_miso[3];
          \current_word$next [4] = io_miso[4];
          \current_word$next [5] = io_miso[5];
          \current_word$next [6] = io_miso[6];
          \current_word$next [7] = io_miso[7];
        end                  
      2'h2:            /* SPI_x4 */      
        begin                     
          \current_word$next [4] = io_miso[0];
          \current_word$next [5] = io_miso[1];
          \current_word$next [6] = io_miso[2];
          \current_word$next [7] = io_miso[3];
        end
      2'h3:            /* SPI_x2 */      
        begin    
          \current_word$next [6] = io_miso[0];
          \current_word$next [7] = io_miso[1];
        end                       
    endcase               
    casez (rst)                               
      1'h1:            
          \current_word$next  = 8'h00;
    endcase
  end   
```

Using something like defines;
```verilog
`define IDLE   2'h0
`define SPI_x8   2'h1
`define SPI_x4   2'h2
`define SPI_x2   2'h3

  always @* begin
    \current_word$next  = current_word;
    casez (fsm_state)
      IDLE:
          /* empty */;
      SPI_x8:
        begin          
          \current_word$next [0] = io_miso[0];
          \current_word$next [1] = io_miso[1];
          \current_word$next [2] = io_miso[2];
          \current_word$next [3] = io_miso[3];
          \current_word$next [4] = io_miso[4];
          \current_word$next [5] = io_miso[5];
          \current_word$next [6] = io_miso[6];
          \current_word$next [7] = io_miso[7];
        end                  
      SPI_x4:
        begin                     
          \current_word$next [4] = io_miso[0];
          \current_word$next [5] = io_miso[1];
          \current_word$next [6] = io_miso[2];
          \current_word$next [7] = io_miso[3];
        end
      SPI_x2:
        begin    
          \current_word$next [6] = io_miso[0];
          \current_word$next [7] = io_miso[1];
        end                       
    endcase               
    casez (rst)                               
      1'h1:            
          \current_word$next  = 8'h00;
    endcase
  end   
```

whitequark, 2019-07-07:
Neither comments nor defines are possible because Yosys' `write_verilog` cannot emit them arbitrarily. It's not even possible to add an attribute to an inner `case`, although it can be done on the `always` block.

I think that without Yosys modifications (which will probably have a hard time being accepted upstream) it's only really possible to set an attribute on the state signal, but this doesn't help a whole lot.

whitequark, 2019-07-07:
@mithro Clifford is on board with this change.

mithro, 2019-07-08:
Awesome! I look forward to seeing it work.

whitequark, 2019-07-09:
The Yosys part is done, and it will be in Yosys 0.9 as well.

mithro, 2019-07-09:
@whitequark - You truly are my hero!

commit 37f363e3 (2019-07-10): back.rtlil: add decodings to cases when switching on a signal.

Fixes #134.
closed 2019-07-10 by whitequark