m-labs/nmigen #135: Support setting FSM encoding attribute

mithro, 2019-07-07:
To get make sure your FSM state variable stays as a specific encoding, most synthesis tools support setting an encoding attribute on the state register. This appears to be something that Synopsys Synplify added and then everyone else copied for compatibility (Quartus II + Libero + Diamond).

```verilog
reg [7:0] current_state /* synthesis syn_encoding="safe,gray" */;
```
```verilog
(* syn_encoding = "safe" *) reg [2:0] my_fsm;
```

A list of the valid encoding schemes seems to be;
 * binary,
 * onehot,
 * twohot,
 * sequential,
 * gray,
 * random,
 * original,

There also seems to be an extra `safe` value which can be added;
> The Safe State Machine option and corresponding syn_encoding
> attribute value safe specify that the software should insert extra logic to
> detect an illegal state and force the state machine’s transition to the reset
> state.

It would be good to be able to do this with nMigen.

whitequark, 2019-07-07:
I'm not yet convinced nMigen needs to provide an abstraction for this. It seems to be a niche feature with more potential for misuse than use, like `full_case`. I've played with Yosys' FSM passes and each time I got them to work on my code, the FSM became bigger and slower...

If you're fine adding platform-specific attributes, you can already do this with:
```python
with m.FSM() as fsm:
    fsm.state.attrs["syn_encoding"] = "onehot"
```

mithro, 2019-07-07:
FYI - Here is what Vivado says about FSM state encoding;

> Vivado synthesis attempts to select the best-suited encoding method for a given FSM.
>
> ## One-Hot State Encoding
> One-Hot State encoding has the following attributes:
> • Is the default encoding scheme for state machine up to 32 states.
> • Is usually a good choice for optimizing speed or reducing power dissipation.
> • Assigns a distinct bit of code to each FSM state.
> • Implements the State Register with one flip-flop for each state.
> • In a given clock cycle during operation, only one bit of the State Register is asserted.
> • Only two bits toggle during a transition between two states.
> 
> ## Gray State Encoding
> Gray State encoding has the following attributes:
> • Guarantees that only one bit switches between two consecutive states.
> • Is appropriate for controllers exhibiting long paths without branching.
> • Minimizes hazards and glitches.
> • Can be used to minimize power dissipation.
>
> ## Johnson State Encoding
> Johnson State encoding is beneficial when using state machines containing long paths with
> no branching (as in Gray State Encoding).
>
> ## Sequential State Encoding
> Sequential State encoding has the following attributes:
> • Identifies long paths
> • Applies successive radix two codes to the states on these paths.
> • Minimizes next state equations.

It may be that it is impossible to know the *best* FSM state encoding until you have timing information?

whitequark, 2019-07-10:
I think for now, the above `fsm.state.attrs["encoding"] = "..."` solution is sufficient since the attribute and its effects are very toolchain specific; and moreover, the toolchain defaults seem to be quite reasonable. If there is good motivation, we can always add an abstraction for this later.

closed 2019-07-10 by whitequark