m-labs/nmigen #136: nextpnr crash: "terminate called after throwing an instance of 'std::out_of_range'"

sbourdeauducq, 2019-07-07:
I'm not sure is this is actually related to nMigen or if this is purely a nextpnr bug, but HeavyX simplesoc_ecp5 no longer works after attempting to use ``nmigen.build``.
I had to update Yosys and nextpnr to get the new I/O system to work, otherwise Yosys would complain about an unknown cell ``IB``.
Now using Yosys ff958830d1097b9bfa3c3b34094e671741ef563d and nexpnr 8455d1f4ffb942c802b65e20748e54a123e08df0.

Repro (without Nix) and error:
```
$ python simplesoc_ecp5.py simplesoc_ecp5.py test
terminate called after throwing an instance of 'std::out_of_range'
  what():  _Map_base::at
build_top.sh: line 4:  7557 Aborted                 nextpnr-ecp5 --quiet --log top.tim --um-45k --package CABGA381 --speed 8 --json top.json --lpf top.lpf --textcfg top.config
```

whitequark, 2019-07-07:
> I had to update Yosys and nextpnr to get the new I/O system to work, otherwise Yosys would complain about an unknown cell `IB`.

Yes, this is expected. Until we release 0.1, you should always run nMigen against master branch of Yosys and nextpnr. The next version of Yosys, 0.9, is the current target, and I'll make sure every patch required for functionality is merged there.

> Repro (without Nix)

Did you attach the repro?

sbourdeauducq, 2019-07-07:
https://git.m-labs.hk/M-Labs/HeavyX/src/branch/master/examples/helloworld_ecp5.py
You only need to add minerva and jtagtap from lambdaconcept in addition to the usual packages.

whitequark, 2019-07-07:
Can you zip the build directory and attach it here?

sbourdeauducq, 2019-07-07:
[repro.zip](https://github.com/m-labs/nmigen/files/3365225/repro.zip)

whitequark, 2019-07-07:
That's a bug in nextpnr, I believe.

whitequark, 2019-07-07:
See https://github.com/YosysHQ/nextpnr/issues/301.

whitequark, 2019-07-07:
@sbourdeauducq Looks like Nix breaks Versioneer:

    # Automatically generated by nMigen 0+unknown. Do not edit.

Can you fix that?

sbourdeauducq, 2019-07-07:
Yeah, that's because ``.git`` is not present in the downloaded sources. It breaks in exactly the same way if you use the "Download ZIP" link on GitHub to get nmigen.

whitequark, 2019-07-07:
> It breaks in exactly the same way if you use the "Download ZIP" link on GitHub to get nmigen.

It does not, which you can check if you want. The reason is the gitattributes file, which causes the following block in `nmigen/_version.py`:
```python
def get_keywords():
    """Get the keywords needed to look up the version information."""
    # these strings will be replaced by git during git-archive.
    # setup.py/versioneer.py will grep for the variable names, so they must
    # each be defined on a line of their own. _version.py will just call
    # get_keywords().
    git_refnames = "$Format:%d$"
    git_full = "$Format:%H$"
    git_date = "$Format:%ci$"
    keywords = {"refnames": git_refnames, "full": git_full, "date": git_date}
    return keywords
```
to be turned into:
```python
def get_keywords():
    """Get the keywords needed to look up the version information."""
    # these strings will be replaced by git during git-archive.
    # setup.py/versioneer.py will grep for the variable names, so they must
    # each be defined on a line of their own. _version.py will just call
    # get_keywords().
    git_refnames = " (HEAD -> master)"
    git_full = "da1f58b7aef99912b32b65fe32cd72b61fd5dc34"
    git_date = "2019-07-07 01:03:59 +0000"
    keywords = {"refnames": git_refnames, "full": git_full, "date": git_date}
    return keywords
```

So nix should do the same thing then.

closed 2019-07-07 by whitequark

sbourdeauducq, 2019-07-07:
Interesting, but even though this replacement does happen correctly in both cases, versionneer still breaks on GitHub archives downloaded and extracted manually and those in the Nix store.
```
nmigen-master> python
>>> import nmigen
>>> nmigen
<module 'nmigen' from '/home/sb/Downloads/nm/nmigen-master/nmigen/__init__.py'>
>>> nmigen.__version__
'0+unknown'
```

```
5x2c0kh722dncpbdlyipjxzd3f2kjj2w-source> python
>>> import nmigen
>>> nmigen
<module 'nmigen' from '/nix/store/5x2c0kh722dncpbdlyipjxzd3f2kjj2w-source/nmigen/__init__.py'>
>>> nmigen.__version__
'0+unknown'
```

whitequark, 2019-07-07:
Ah, sorry, my bad. I'll take a look at it myself then.

sbourdeauducq, 2019-07-07:
This is also affecting ARTIQ (and leaving ``.git`` around with Nix is actually annoying), and I didn't know about this gitattributes thing so I ended up bypassing versioneer.

commit cb02a452 (2019-07-07): vendor.lattice_ecp5: don't leave LUT inputs disconnected.

This causes YosysHQ/nextpnr#301.

Fixes #136.