m-labs/nmigen #141: tracer.get_var_name returns None instead of default on pypy3

programmerjake, 2019-07-09:
on pypy3 (pypy 7.2.0-beta0),
`tracer.get_var_name` returns `None` when called through `Signal.__init__` as follows:

```python
from nmigen import Signal
v = Signal(reset_less=False)
print(v.name)
```

instead of returning `None`, it would be better to return `default` or raise an exception.

having `None` as the name causes `Simulator` to crash.

whitequark, 2019-07-09:
I don't think that's the right solution because this way, with PyPy, you cannot generate meaningful Verilog at all. You'd think simulation would be much faster on PyPy, but I've evaluated it in the past (with a workaround for this bug), and it is *significantly* slower than on CPython.

Since PyPy's bytecode is quite similar to CPython's, the right solution here is to not hack around it but to implement proper support for PyPy bytecode. Because of its slowness I do not think it currently makes any sense to put effort into it, but if you are interested in PyPy I would accept a patch that does that and add it to Travis' build matrix.

closed 2019-07-09 by whitequark

programmerjake, 2019-07-10:
> You'd think simulation would be much faster on PyPy, but I've evaluated it in the past (with a workaround for this issue), and it is _significantly_ slower than on CPython.

I've found pypy to be much slower at elaborating but much faster at simulation. As an example, running [test_core.py in ieee754fpu](https://git.libre-riscv.org/?p=ieee754fpu.git;a=blob;f=src/ieee754/div_rem_sqrt_rsqrt/test_core.py;h=a9a3d29fed42cfe5d37665b1346dbf1dfc4e8925;hb=4a74b6d3bb4dca7d6c855a9b9348d9d40dec788b), which tests part of a div/rem/fdiv/fsqrt/frsqrt pipeline, it takes about 930s using pypy3 (7.2.0-beta0) and 1370s using python3 (3.6.8) on my Ryzen 2600. It tests about 140k simulated clock cycles.

whitequark, 2019-07-10:
Interesting. I think the reason here is that the workloads are different. IIRC I benchmarked simulation on small cores, where it might be that the tracing JIT doesn't work as well.

Anyway, that's good to know. In light of this information I think it is worthwhile to support PyPy as a first-class runtime.

PR #142 (2019-07-09): allow tracer.get_var_name to work on pypy3
Fixes #141
commit 62b3e366 (2019-07-09): tracer: add PyPy support to get_var_name().

Fixes #141.