m-labs/nmigen #147: Reduce-op (reduce-or, reduce-and, reduce-xor)

whitequark, 2019-07-14:
Verilog has these and they can be quite useful in bit-twiddling code, potentially reducing bugs as doing the same thing explicitly has a potential to introduce them. (I'm pretty sure I did introduce bugs like that.) Also, we already have `Value.bool`, which is really a reduce-or.

Bikeshed: how should they be called? I think `Value.reduce_or`, etc are OK. But another option is `Value.any_bits` for reduce-or, `Value.all_bits` for reduce-and, `Value.even_bits` for reduce-xor, and `Value.odd_bits` for reduce-xnor.

Bikeshed: should we even keep `Value.bool`? Does everyone agree that "any bit set" is the appropriate semantics for converting any value to a 1-bit one? Migen didn't have it so it's not even a backwards compatibility hazard.

whitequark, 2019-07-14:
We probably want to keep `Value.bool` for two reasons: first, the Yosys `$reduce_bool` cell has the same semantics but different interpretation as `$reduce_or` cell (so it can be useful when roundtripping back to source), and second, Verilog does in fact define the mux `?:`, conditional `if` and logical not `!` semantics consistent with `Value.bool`.

sbourdeauducq, 2019-07-14:
Do we need those functions? It's easy enough to do it already with the Python ``operator`` module and the ``reduce`` function from ``functools``.

whitequark, 2019-07-14:
@sbourdeauducq You can't do xnor that way, it's somewhat annoying to write, and it results in significantly less readable Verilog output / more bloated RTLIL, with one cell per bit.

whitequark, 2019-09-13:
`any` and `all` are in b23a979.

whitequark, 2019-09-13:
@programmerjake I agree with your argument, any method with "parity" in it seems bound to be inherently ambiguous. As much as I dislike it, I think adding just `.xor()` alongside `.any()` and `.all()` is the only reasonable solution to this.

whitequark, 2019-09-13:
As an example of what I mean by inherently ambiguous, @emilazy suggests calling it `.parity()` where `0` means even and `1` means odd (which is reasonable in principle--it's popcnt mod 2, and the evenness of parity matches evenness of the number), but... x86 does the inverse of that for its parity flag.

commit 0a00d533 (2019-09-13): hdl.ast: add VAlue.xor, mapping to $reduce_xor.

Fixes #147.
commit 32310aec (2019-09-13): hdl.ast: add Value.xor, mapping to $reduce_xor.

Fixes #147.
closed 2019-09-13 by whitequark