m-labs/nmigen #151: Question: How to change FIFO clock dommains?

andresdemski, 2019-07-14:
Hello
I'm starting using nMigen and i want instantiate the AsyncFifoBuffered library. I found that the fifo clock domain are `read` and  `write`, and i need to connect them to domains called `input` and `output`. 

* Is there any easy way to make this connection? 
* Should libraries have an argument to specify the domain names? 

Thx.
Andres

peteut, 2019-07-14:
@andresdemski you are looking for 

```python
m.submodules += DomainRenamer({"read": "input", "write": "output"})(<your-module>)
```

andresdemski, 2019-07-15:
With this approach, you always need to know the implementation (or it needs to be well documented). A domain argument in libraries should be a good idea.

whitequark, 2019-07-15:
@andresdemski I entirely agree, #97 is the tracking issue for adding an explicit argument.

andresdemski, 2019-07-16:
Ok, Thanks you!

closed 2019-07-16 by andresdemski