m-labs/nmigen #153: support for memories with N read-only ports + 1 write-only port?

sbourdeauducq, 2019-07-17:
One trick I've used several times to implement a memory with several (typically N=2...4) read ports plus one write port is to use N dual-port memories with, for each memory, one port used for reading and one port used for writing. The individual write ports are connected in parallel to form the write port into what is the equivalent of a N+1-port memory with N read-only port and 1 write-only port.

Should nMigen automatically implement this pattern when applicable, or should this be done manually by the user?

whitequark, 2019-07-17:
You can already do this: simply request the appropriate amount of read ports by calling `mem.read_port()` several times. The toolchain should duplicate the memories to implement it; I know Yosys does, but I haven't checked ISE, Vivado and Diamond. If they do not it would be an issue.

sbourdeauducq, 2019-07-24:
Hmm. One would have to check that carefully (I remember problems with ISE a long time ago), though this would be one of many issues that would disappear if we let Yosys synthesize memories for all FPGA families.

whitequark, 2019-07-24:
According to IRC conversation with @cr1901, ISE and Vivado can do that.

cr1901, 2019-07-24:
Rationale: The default config file of `lm32` (including the one used by MiSoC) relies on the synthesizer to duplicate memories for the reg file. I've built enough `lm32` SoCs in the past on ISE and Vivado successfully in that configuration to know that the toolchains will duplicate memories.

whitequark, 2019-07-31:
Closing; please reopen if there are actual toolchain bugs we need to work around.

closed 2019-07-31 by whitequark