m-labs/nmigen #157: Submodule named 'ddr2_0__a' already exists after  698b005 with Atlys

Fatsie, 2019-07-20:
After yesterdays commit  698b005 I get following error when using ddr2.a from AtlysPlatform (pull-request m-labs/nmigen-boards#15).
Python code:
```python
#!/bin/env python3
from nmigen import *
from nmigen.build import *
from nmigen_boards.atlys import AtlysPlatform

class subsignal_bug(Elaboratable):
    def __init__(self, clk_name="clk"):
        self._clk_name = clk_name

    def elaborate(self, platform):
        m = Module()

        clk = platform.request(self._clk_name)
        m.domains.sync = ClockDomain()
        m.d.comb += ClockSignal().eq(clk.i)

        ddr2 = platform.request("ddr2")
        m.d.comb += ddr2.a.o.eq(Const(5,13))

        return m

p = AtlysPlatform()
test = subsignal_bug(clk_name="clk100")

products = p.build(test)
```
And this is the trace-back:
```
(nmigen) [verhaegs@localhost nmigen]$ ./atlys_bug.py 
Traceback (most recent call last):
  File "./atlys_bug.py", line 27, in <module>
    products = p.build(test)
  File "/home/verhaegs/anaconda2/envs/nmigen/lib/python3.7/site-packages/nmigen/build/plat.py", line 47, in build
    plan = self.prepare(fragment, name, **kwargs)
  File "/home/verhaegs/anaconda2/envs/nmigen/lib/python3.7/site-packages/nmigen/build/plat.py", line 73, in prepare
    add_pin_fragment(pin, self.get_output(pin, port, attrs, invert))
  File "/home/verhaegs/anaconda2/envs/nmigen/lib/python3.7/site-packages/nmigen/vendor/xilinx_spartan_3_6.py", line 300, in get_output
    o_O=port[bit]
  File "/home/verhaegs/anaconda2/envs/nmigen/lib/python3.7/site-packages/nmigen/hdl/dsl.py", line 88, in __setitem__
    return self.__setattr__(name, value)
  File "/home/verhaegs/anaconda2/envs/nmigen/lib/python3.7/site-packages/nmigen/hdl/dsl.py", line 85, in __setattr__
    self._builder._add_submodule(submodule, name)
  File "/home/verhaegs/anaconda2/envs/nmigen/lib/python3.7/site-packages/nmigen/hdl/dsl.py", line 432, in _add_submodule
    raise NameError("Submodule named '{}' already exists".format(name))
NameError: Submodule named 'ddr2_0__a' already exists
```
 I tried to reproduce it with (sub)signals in other platform but did not succeed. Reverting 698b005 removes the problem.

whitequark, 2019-07-20:
cc @nakengelhardt

nakengelhardt, 2019-07-20:
Ah, yes, the vendor files do a lot of 
```
for bit in range(len(port)):
    m.submodules[pin.name] = Instance(...)
```
which is no longer allowed.

PR #158 (2019-07-21): remove duplicate iobuf submodule names
Fixes #157.

Vendor files create a lot of IObufs with duplicate names:
```for bit in range(len(port)):
    m.submodules[pin.name] = Instance(...)
```
which is no longer allowed as of 698b005.

This appends a suffix for each bit:
```
for bit in range(len(p_port)):
    m.submodules["{}_{}".format(pin.name, bit)] = Instance(...)
```
closed 2019-07-21 by whitequark