m-labs/nmigen #169: Platform default clock

zignig, 2019-08-05:
Updated  to the latest nmigen/nimgen-boards  and the defined clock for my tinybx platform stopped working.

The following code 

```
def elaborate(self, platform):
        clk16 = platform.request("clk16", 0)

        m = Module()
        m.domains.sync = ClockDomain()
        m.d.comb += ClockSignal().eq(clk16.i)

        m.submodules.boneless = self.b
        return m
```
returns  a long error ending in

```
  File "/usr/local/lib/python3.6/dist-packages/nmigen/hdl/ir.py", line 305, in _propagate_domains_up
    subfrag._propagate_domains_up(hierarchy + (hier_name,))
  File "/usr/local/lib/python3.6/dist-packages/nmigen/hdl/ir.py", line 341, in _propagate_domains_up
    self.add_domains(subfrag.domains[domain])
  File "/usr/local/lib/python3.6/dist-packages/nmigen/hdl/ir.py", line 140, in add_domains
    assert domain.name not in self.domains
AssertionError
```
Tried removing the clocking all together to try to get the new default_clk to connect , same issue.

Bisected it down to https://github.com/m-labs/nmigen/commit/8854ca03ae01811bf9e2b90aa58a301d6cbbd58c , not sure why it's not getting the default clock, or taking the existing one.

whitequark, 2019-08-05:
Yes, known bug. I'll fix it soon.

Fatsie, 2019-08-10:
I have same assertion but caused by use of hierarchical modules. Reporting as extra test case:
```python
#!/bin/env python3
from nmigen import *
from nmigen.back import rtlil

from nmigen_boards.icebreaker import ICEBreakerPlatform

class SubMod(Elaboratable):
    def __init__(self):
        self.i = Signal()
        self.o = Signal()

    def elaborate(self, platform):
        m = Module()

        m.d.sync += self.o.eq(self.i)

        return m

class Top(Elaboratable):
    def elaborate(self, platform):
        m = Module()

        sw = platform.request("user_btn")
        led = platform.request("user_led")

        m.submodules.submod = submod = SubMod()

        m.d.comb += [
            submod.i.eq(sw.i),
            led.o.eq(submod.o),
        ]

        return m

plan = ICEBreakerPlatform().build(Top(), do_build=False)
```

commit ada1d6a6 (2019-08-19): build.plat: do not prepare fragments twice.

Fixes #169.
closed 2019-08-19 by whitequark