m-labs/nmigen #184: ASIC support tracking issue

whitequark, 2019-08-22:
A number of people have expressed a desire to use nMigen for ASIC design. This issue tracks missing aspects of that workflow.

 - [ ] On FPGAs, resets are generally positive polarity, and using the wrong polarity can have severe consequences for timing. On ASICs, resets are generally negative polarity, such that the chip is held in reset during power rail ramp-up.

(There are likely more.)

programmerjake, 2019-08-22:
I would think that most SRAMs don't get reset on ASICs either, so that shouldn't interfere with BRAM inference.

whitequark, 2019-08-23:
> I would think that most SRAMs don't get reset on ASICs either, so that shouldn't interfere with BRAM inference.

The problem isn't that the SRAM itself gets reset, but that you might want to fold address or data registers into an SRAM core. This is less of a problem in nMigen than in Verilog because some registers are implicit in the read port, but it's still a problem if a few of them are pipelined; e.g. Xilinx offers this as an option on BRAMs.

programmerjake, 2019-08-23:
Ah, missed that.