m-labs/nmigen #192: yosys v0.8 not supported, causes assertion error

RobertBaruch, 2019-09-02:
When attempting to generate from the `ctr.py` example:

```
$ python3 ctr.py generate
Traceback (most recent call last):
  File "ctr.py", line 19, in <module>
    main(ctr, ports=[ctr.o])
  File "/home/robertbaruch/.local/lib/python3.6/site-packages/nmigen/cli.py", line 76, in main
    main_runner(parser, parser.parse_args(), *args, **kwargs)
  File "/home/robertbaruch/.local/lib/python3.6/site-packages/nmigen/cli.py", line 58, in main_runner
    output = verilog.convert(fragment, name=name, ports=ports)
  File "/home/robertbaruch/.local/lib/python3.6/site-packages/nmigen/back/verilog.py", line 68, in convert
    return _convert_il_text(il_text, strip_src)
  File "/home/robertbaruch/.local/lib/python3.6/site-packages/nmigen/back/verilog.py", line 27, in _convert_il_text
    raise YosysError("Yosys %d.%d is not suppored", *version)
nmigen.back.verilog.YosysError: ('Yosys %d.%d is not suppored', 0, 8)

$ yosys -V
Yosys 0.8+652 (git sha1 d16178f2, clang 6.0.0-1ubuntu2 -fPIC -Os)
```

whitequark, 2019-09-02:
That is correct. Yosys 0.8 is not supported. Use 0.9 or master.

RobertBaruch, 2019-09-02:
Oh wait, I was using master that I pull on August 15 -- it's been updated since, apparently!

closed 2019-09-02 by RobertBaruch