m-labs/nmigen #196: Signal's `max` parameter isn't a max

RobertBaruch, 2019-09-05:
According to the docs:

```
max : int or None
        If ``shape`` is ``None``, the signal bit width and signedness are
        determined by the integer range given by ``min`` (inclusive,
        defaults to 0) and ``max`` (exclusive, defaults to 2).
```

But the maximum value of a signal is by definition inclusive for discrete ranges. I'd prefer that `max` be changed to inclusive, defaults to 1, and all uses of `max` instead use `max+1`.

whitequark, 2019-09-05:
> I'd prefer that `max` be changed to inclusive, defaults to 1, and all uses of `max` instead use `max+1`.

This isn't possible. Imagine a situation where code is ported from oMigen to nMigen. This means that, for some time, code in nMigen and code in nMigen compat mode (which behaves ~exactly like oMigen) must coexist. Consider this situation: would you like to have to scroll up and see which `Signal` is imported to figure out what `max` means?

Also, there's alerady nontrivial amount of nMigen code, so changing this now would be breaking all of it, and moreover, as you see in README, the API of `nmigen.hdl` is already stable. But even if this wasn't the case, the cause listed above is already prohibitive.

However, what *is* possible is deprecating `min`/`max` that have this confusing semantics, and instead adding a `Signal.range()` constructor, which behaves in a clear way: if you create a signal with `Signal.range(x, [y])`, every `v for v in range(x, [y])` is a valid value for this signal.

/cc @emilazy

sbourdeauducq, 2019-09-05:
Those semantics are the same as Python's ``range`` and list slices, which is why I chose them for Migen initially.

whitequark, 2019-09-05:
Yes. The problem isn't so much semantics as the name; it says `max` whereas it's not the maximum value. Python calls these values `start` and `stop`.

sbourdeauducq, 2019-09-05:
Alright, then OK to add and recommend start/stop and still support min/max for legacy reasons.

whitequark, 2019-09-05:
I think it should be a separate constructor, similar to `Signal.like()`. The current signature of `Signal()` is very strange, with several mutually exclusive positional and keyword arguments.

RobertBaruch, 2019-09-05:
Yeah, changing the meaning of max it's problematic and would break backward compatibility. I'm in favor of a new constructor/builder.

emilazy, 2019-09-05:
Agreed, I'd much rather Signal had a few constructors with simple signatures than the current monolith.

commit ccfbccc0 (2019-09-08): hdl.ast: add Signal.range(...), to replace Signal(min=..., max=...).

Fixes #196.
closed 2019-09-08 by whitequark