m-labs/nmigen #197: Consider adding Default to Module in dsl.py

RobertBaruch, 2019-09-08:
It seems to be more consistent with most other commonly-known languages to have the syntax for a default case look like this:
```
with m.Switch(thing):
  with m.Case(1):
    ...
  with m.Case(2):
    ...
  with m.Default():
    ...
```

than this:
```
with m.Switch(thing):
  with m.Case(1):
    ...
  with m.Case(2):
    ...
  with m.Case("default"):
    ...
```

whitequark, 2019-09-08:
Hm, makes sense.

whitequark, 2019-09-08:
Actually, hang on.

`with m.Case("default"):` is not valid in nMigen (this seems to be a mix of syntax from oMigen and nMigen), what you can do is `with m.Case():`.

programmerjake, 2019-09-08:
if `with m.Case():` is going to be the way to get a default case, that needs to be documented somewhere, I was not aware that nmigen supported default cases at all.

I personally prefer having a separate `Default` function, since that is more explicit.

whitequark, 2019-09-08:
I think having a `Default()` function as an alias for `Case()` is fine.

closed 2019-09-08 by whitequark

RobertBaruch, 2019-09-08:
Ah, ok. I had a suspicion that default wasn't actually supported in nMigen (since I couldn't find `"default"` in the source) so I fell back to the Migen docs. So now you fixed two issues :D

commit 3f6abc0b (2019-09-08): hdl.dsl: add Default(), an alias for Case() with no arguments.

Fixes #197.