m-labs/nmigen #207: First-class enumerated signals

emilazy, 2019-09-14:
#206:
> we don't currently have anything like first-class enumerated signals

whitequark, 2019-09-14:
I'm not actually sure how to add these. You want to be able to do something like `s == "FOO"` but then they can't be `Value`s without some real awful coercion going on.

emilazy, 2019-09-14:
Please don't use strings: https://docs.python.org/3/library/enum.html

whitequark, 2019-09-14:
OK, but if you define an `IntEnum`, you can use regular `Signal`s. And also it would not help at all the FSM use case.

emilazy, 2019-09-14:
wish you had type inference of extensible variants ¯\_(ツ)_/¯

whitequark, 2019-09-14:
Wrong language for that.

RobertBaruch, 2019-09-14:
Here are my thoughts:

```
from nmigen import *
from nmigen.cli import main
from nmigen.asserts import *
from enum import Enum, unique

# Two different types of "enums".

class Thing(object):
    """A fake enumerated value.

    This isn't actually an Enum, but is just a bag of constants,
    and not at all Pythonic.
    """
    DOG = 0
    CAT = 1
    SQUIRREL = 2

    @classmethod
    def signal(cls):
        """Returns a Signal with the right number of bits for the Enum.

        src_loc_at tells nMigen which stack frame to get the variable
        name from, so the one above this one.
        """
        return Signal.range(0, 3, src_loc_at=1)

@unique
class ConstingEnumThing(Enum):
    """A real Enum.

    In general, we shouldn't care what the actual value of an enum
    is. If you do, you probably don't want an enum but a Const.

    We'll see later that the enum values don't matter. You can auto() them,
    or even use strings.
    """
    DOG = 0
    CAT = 1
    SQUIRREL = 2

    @classmethod
    def signal(cls):
        """Returns a Signal with the right number of bits for the Enum.

        src_loc_at tells nMigen which stack frame to get the variable
        name from, so the one above this one.
        """
        return Signal.range(0, 3, src_loc_at=1)

    @property
    def const(self):
        """Returns a Const with the right number of bits for the Enum.

        Although we use self.value, we'll see later on that we can
        dispense with that.
        """
        return Const(self.value, ConstingEnumThing.signal().shape())

class CatDetector(Elaboratable):
    """Uses Thing, which isn't an Enum.

    This issues a warning because width(Thing.CAT) = 1, and
    width(Thing.DOG) = 2.
    """

    def __init__(self):
        self.input = Thing.signal()
[... 193 lines cut ...]
    m.submodules.meow = meow
    # m.submodules.meow2 = meow2
    m.submodules.meow3 = meow3
    m.submodules.meow4 = meow4
    m.submodules.meow5 = meow5
    m.submodules.meow6 = meow6

    main(
        m,
        ports=[
            meow.input,
            meow.output,
            # meow2.input, meow2.output,
            meow3.input,
            meow3.output,
            meow4.input,
            meow4.output,
            meow5.input,
            meow5.output,
            meow6.input,
            meow6.output1,
            meow6.output2,
            meow6.output3
        ])
```

whitequark, 2019-09-14:
Yeah, it's challenging to implement. Have you considered using an `enum.IntEnum`? That would work out of the box with nMigen.

RobertBaruch, 2019-09-14:
IntEnum is [frowned upon](https://docs.python.org/3/library/enum.html), mainly because it can be compared to other IntEnums (which enums shouldn't be) and other values in the same IntEnum, which is fine if you want that sort of thing.

> For the majority of new code, Enum and Flag are strongly recommended, since IntEnum and IntFlag break some semantic promises of an enumeration (by being comparable to integers, and thus by transitivity to other unrelated enumerations). IntEnum and IntFlag should be used only in cases where Enum and Flag will not do; for example, when integer constants are replaced with enumerations, or for interoperability with other systems.

I'd prefer Enums over IntEnums where I need a signal whose values are orthogonal with all other values, can be compared for equality and assigned, but have no other defined numeric representation (except under implementation).

whitequark, 2019-09-14:
The way normal `Enum` works is at odds with nMigen's semantics, and I don't see any straightforward way to resolve that, so you're stuck working with the "frowned upon" mechanism right now.

whitequark, 2019-09-14:
Hmm... maybe `Value.wrap` could be taught to recognize `enum.Enum`?

RobertBaruch, 2019-09-14:
IntEnum seems to have the "you just have to know when to use .const" problem...

```python
class IntThing(IntEnum):
    """An enumerated value based on IntEnum.
    """
    DOG = 0
    CAT = 1
    SQUIRREL = 2

    @classmethod
    def signal(cls):
        """Returns a Signal with the right number of bits for the Enum.

        src_loc_at tells nMigen which stack frame to get the variable
        name from, so the one above this one.
        """
        return Signal.range(0, 3, src_loc_at=1)

    @property
    def const(self):
        """Returns a Const with the right number of bits for the Enum.

        Although we use self.value, we'll see later on that we can
        dispense with that.
        """
        return Const(self.value, ConstingEnumThing.signal().shape())

class IntEnumCatDetector(Elaboratable):
    """CatDetector using IntEnum.

    You can dispense with .value, but you must still use .const.
    """

    def __init__(self):
        self.input = IntThing.signal()
        self.output = Signal()

    def elaborate(self, platform):
        m = Module()
        self.detectCat(m, IntThing.CAT.const)
        return m

    def detectCat(self, m, comparand):
        with m.Switch(comparand):
            with m.Case(IntThing.DOG, IntThing.SQUIRREL):
                m.d.comb += self.output.eq(0)
            with m.Default():
                m.d.comb += self.output.eq(1)
```

```
elab.py:300: SyntaxWarning: Case value '10' is wider than test (which has width 1); comparison will never be true
  with m.Case(IntThing.DOG, IntThing.SQUIRREL):
```

RobertBaruch, 2019-09-14:
> Hmm... maybe `Value.wrap` could be taught to recognize `enum.Enum`?

It could, but then we'll have problems in `Switch/Case` with wideness warnings, no?

whitequark, 2019-09-14:
> It could, but then we'll have problems in `Switch/Case` with wideness warnings, no?

Why? It could recognize `Enum`, check the maximum value, and emit a `Const` of an appropriate size.

RobertBaruch, 2019-09-14:
> > It could, but then we'll have problems in `Switch/Case` with wideness warnings, no?
> 
> Why? It could recognize `Enum`, check the maximum value, and emit a `Const` of an appropriate size.

Yes -- that's the `IdealEnumCatDetector` case in my example. The idea is that `enumToConst` and `enumToValue` are the conversions that would be in nMigen.

whitequark, 2019-09-15:
@RobertBaruch So what do you think about adding (a) coerion from any `Enum` to a `Const` of an appropriate width, and (b) adding `Signal.enum` that creates a normal `Signal` from any `Enum` by interrogating it for minimum/maximum value, and checking that they're indeed integer?

RobertBaruch, 2019-09-15:
I'm a fan! Anything would be better than having to code it up for each project :) As long as it works nicely with at least `Switch/Case` and `If`, I'd be happy. `Signal.enum` would be especially useful, and `Enum`->`Const` as well as `Enum` -> `Value` would be great.

whitequark, 2019-09-15:
Any `Const` is a `Value` already.

RobertBaruch, 2019-09-15:
But then where does this error come from? If `Enum` becomes a `Const`, wouldn't that cause this error?

```
# We can't use the Const version on all the enum values, because:
#
# File "/home/robertbaruch/.local/lib/python3.6/site-packages/nmigen/hdl/dsl.py", line 286, in Case
#    switch_data["cases"][new_values] = self._statements
# TypeError: unhashable type: 'Const'
#
# class EnumConstCatDetector(Elaboratable):
#     def __init__(self):
#         self.input = ConstingEnumThing.signal()
#         self.output = Signal()

#     def elaborate(self, platform):
#         m = Module()
#         self.detectCat(m, ConstingEnumThing.CAT.const)
#         return m

#     def detectCat(self, m, comparand):
#         with m.Switch(comparand):
#             with m.Case(ConstingEnumThing.DOG.const, ConstingEnumThing.const):
#                 m.d.comb += self.output.eq(0)
#             with m.Default():
#                 m.d.comb += self.output.eq(1)
```

whitequark, 2019-09-15:
Ah, right. The matching code would have to be extended to handle `Enum`, too. But that's fine.

RobertBaruch, 2019-09-15:
I can work on this, if you like.

closed 2019-09-16 by whitequark

RobertBaruch, 2019-09-17:
I'm VERY appreciative of this!

[image]

mithro, 2019-09-17:
This is amazing! Great work.

commit 2fca690e (2019-09-16): hdl.{ast,dsl}: add Signal.enum; coerce Enum to Value; accept Enum patters.

Fixes #207.
commit d03921e2 (2019-09-16): hdl.{ast,dsl}: add Signal.enum; coerce Enum to Value; accept Enum patterns.

Fixes #207.
commit 4777a7b3 (2019-09-16): hdl.{ast,dsl}: add Signal.enum; coerce Enum to Value; accept Enum patterns.

Fixes #207.