m-labs/nmigen #208: How do I simulate?

RobertBaruch, 2019-09-15:
I know it has something to do with running `python3 <py file> simulate`, but what is the structure of that py file? Preferably something that doesn't involve `unittest` or `FHDLTestCase`.

whitequark, 2019-09-15:
See `examples/basic/uart.py` for an example of how to simulate without using `nmigen.cli`.

RobertBaruch, 2019-09-15:
Perfect, thanks!

closed 2019-09-15 by RobertBaruch

RobertBaruch, 2019-09-15:
Fun, it works! Although I wish I didn't have to specify the `domain` all the time. Because of my apparently poor choice of clock domain name :P

```
from nmigen import *
from nmigen.asserts import *
from nmigen.hdl.ast import *
from nmigen.back import pysim

from transparent_latch import TransparentLatch

if __name__ == "__main__":
    latch = TransparentLatch(8)
    ports = latch.ports()

    with pysim.Simulator(
            latch,
            vcd_file=open("transparent_latch.vcd", "w"),
            gtkw_file=open("transparent_latch.gtkw", "w"),
            traces=ports) as sim:
        sim.add_clock(1e-6, domain="pos")

        def process():
            # initial reset
            yield latch.input.eq(0)
            yield latch.en.eq(0)
            yield Tick(domain="pos")
            yield Tick(domain="pos")
            yield latch.input.eq(0xFF)
            yield Tick(domain="pos")
            yield latch.en.eq(1)
            yield Tick(domain="pos")
            yield latch.en.eq(0)
            yield Tick(domain="pos")

        # or add_process?
        sim.add_sync_process(process(), domain="pos")
        sim.run()
```

whitequark, 2019-09-15:
You should use `add_sync_process` and `yield`, or `add_process` and `yield Tick(...)`; these are mutually exclusive.