m-labs/nmigen #209: Formatting settings?

RobertBaruch, 2019-09-15:
Does the project use yapf, autopep8, or black for standard formatting? Any special parameters passed to these?

whitequark, 2019-09-15:
I don't use autoformatters, most of them are downright terrible. clang-format is okay but this isn't a C project.

cr1901, 2019-09-15:
@whitequark Thoughts on adding a flake8 step to Travis that checks immediately whether the code is properly formatted (based on the rules you decide- i.e. remove the restrictive 80 char per line limit), and fails if it's not?

whitequark, 2019-09-15:
I'm not sure if there's much point to it.

RobertBaruch, 2019-09-15:
I prefer to leave formatting up to the auto-formatter so that I don't have to do any formatting myself. Specifying the project policy as no auto-formatting, though, is perfectly fine. I'll just have to remember to turn the auto-formatter off when I write code for nMigen :)

whitequark, 2019-09-15:
Yeah, I would prefer no autoformatting for now. Could perhaps be changed in the future.

whitequark, 2019-09-15:
Somewhat annoyingly, `nmigen-boards` *would* benefit from autoformatting, but any Python autoformatter I know of would mess up the board definitions badly.

closed 2019-09-15 by whitequark

RobertBaruch, 2019-09-15:
That's what [`# yapf: disable` and `# yapf: enable`](https://github.com/google/yapf#why-does-yapf-destroy-my-awesome-formatting) are for :)

whitequark, 2019-09-15:
What's the point of using yapf if you disable it on 99% of code?

RobertBaruch, 2019-09-15:
The board files aren't 99% of the code, are they?

whitequark, 2019-09-15:
They are 99% of code in nmigen-boards (which is a different repository with different tooling settings).