m-labs/nmigen #220: Prune Yosys-internal attributes in emitted Verilog

whitequark, 2019-09-21:
Yosys-internal attributes like `init` can confuse vendor tools, with potentially severe consequences. We should remove them.

Ideally we would map `src` to something sensible, but I'm not sure if this is supported by any vendor tools.

ZirconiumX, 2019-09-21:
Can we add `toolchain:quartus` too?

daveshah1, 2019-09-21:
Would need some work in the Verilog backend, but some cases of `src` might be mappable to the `line preprocessor command

whitequark, 2019-09-21:
@ZirconiumX Done

@daveshah1 Is there a `file`?

whitequark, 2019-09-21:
Oh nevermind, it's `\`line <line> "<filename>"`. That would be really nice to have.

daveshah1, 2019-09-21:
`line can take a filename

[image]

whitequark, 2019-09-22:
Ah crap, this is really bad--it's not possible to remove `init` with `attrmap` since it would also erase it from the actual register in the Verilog output.

whitequark, 2019-09-22:
https://github.com/YosysHQ/yosys/pull/1393

whitequark, 2019-09-24:
In ISE, it is not possible to use user-defined attributes for anything. ISE also loudly complains about seeing any `nmigen.*` attributes, as well as `src` attributes. So all of these should be stripped (perhaps leaving around a `top.debug.v` file with the attributes present).

In Vivado, it is possible to use user-defined attributes, and indeed we do use it. Vivado does not complain about user-defined attributes, so there is no need to strip any of them.

whitequark, 2019-09-24:
In Diamond, it appears to not be possible to use user-defined attributes for anything. (`ncd_attr`/`ngd_attr` can only manipulate built-in ones, and only post-synthesis anyway.) Moreover, edif2ngd complains about all of our attributes (probably anything except `syn_*`), so these should be stripped.

ZirconiumX, 2019-09-24:
```
Warning (10335): Unrecognized synthesis attribute "nmigen.hierarchy" at top.v(4) File: //wsl$/Ubuntu-18.04/home/lofty/de_10/build/top.v Line: 4
Warning (10335): Unrecognized synthesis attribute "generator" at top.v(5) File: //wsl$/Ubuntu-18.04/home/lofty/de_10/build/top.v Line: 5
Warning (10335): Unrecognized synthesis attribute "src" at top.v(7) File: //wsl$/Ubuntu-18.04/home/lofty/de_10/build/top.v Line: 7
Warning (10335): Unrecognized synthesis attribute "src" at top.v(9) File: //wsl$/Ubuntu-18.04/home/lofty/de_10/build/top.v Line: 9
Warning (10335): Unrecognized synthesis attribute "src" at top.v(11) File: //wsl$/Ubuntu-18.04/home/lofty/de_10/build/top.v Line: 11
Warning (10335): Unrecognized synthesis attribute "src" at top.v(13) File: //wsl$/Ubuntu-18.04/home/lofty/de_10/build/top.v Line: 13
Warning (10335): Unrecognized synthesis attribute "src" at top.v(15) File: //wsl$/Ubuntu-18.04/home/lofty/de_10/build/top.v Line: 15
Warning (10335): Unrecognized synthesis attribute "src" at top.v(17) File: //wsl$/Ubuntu-18.04/home/lofty/de_10/build/top.v Line: 17
Warning (10335): Unrecognized synthesis attribute "src" at top.v(19) File: //wsl$/Ubuntu-18.04/home/lofty/de_10/build/top.v Line: 19
Warning (10335): Unrecognized synthesis attribute "src" at top.v(21) File: //wsl$/Ubuntu-18.04/home/lofty/de_10/build/top.v Line: 21
Warning (10335): Unrecognized synthesis attribute "src" at top.v(23) File: //wsl$/Ubuntu-18.04/home/lofty/de_10/build/top.v Line: 23
Warning (10335): Unrecognized synthesis attribute "src" at top.v(25) File: //wsl$/Ubuntu-18.04/home/lofty/de_10/build/top.v Line: 25
Warning (10335): Unrecognized synthesis attribute "src" at top.v(27) File: //wsl$/Ubuntu-18.04/home/lofty/de_10/build/top.v Line: 27
Warning (10335): Unrecognized synthesis attribute "src" at top.v(29) File: //wsl$/Ubuntu-18.04/home/lofty/de_10/build/top.v Line: 29
Warning (10335): Unrecognized synthesis attribute "src" at top.v(31) File: //wsl$/Ubuntu-18.04/home/lofty/de_10/build/top.v Line: 31
Warning (10335): Unrecognized synthesis attribute "src" at top.v(33) File: //wsl$/Ubuntu-18.04/home/lofty/de_10/build/top.v Line: 33
Warning (10335): Unrecognized synthesis attribute "src" at top.v(35) File: //wsl$/Ubuntu-18.04/home/lofty/de_10/build/top.v Line: 35
Warning (10335): Unrecognized synthesis attribute "src" at top.v(37) File: //wsl$/Ubuntu-18.04/home/lofty/de_10/build/top.v Line: 37
Warning (10335): Unrecognized synthesis attribute "src" at top.v(39) File: //wsl$/Ubuntu-18.04/home/lofty/de_10/build/top.v Line: 39
Warning (10335): Unrecognized synthesis attribute "src" at top.v(41) File: //wsl$/Ubuntu-18.04/home/lofty/de_10/build/top.v Line: 41
Warning (10335): Unrecognized synthesis attribute "src" at top.v(43) File: //wsl$/Ubuntu-18.04/home/lofty/de_10/build/top.v Line: 43
Warning (10335): Unrecognized synthesis attribute "src" at top.v(45) File: //wsl$/Ubuntu-18.04/home/lofty/de_10/build/top.v Line: 45
Warning (10335): Unrecognized synthesis attribute "src" at top.v(47) File: //wsl$/Ubuntu-18.04/home/lofty/de_10/build/top.v Line: 47
Warning (10335): Unrecognized synthesis attribute "src" at top.v(50) File: //wsl$/Ubuntu-18.04/home/lofty/de_10/build/top.v Line: 50
Warning (10335): Unrecognized synthesis attribute "src" at top.v(52) File: //wsl$/Ubuntu-18.04/home/lofty/de_10/build/top.v Line: 52
Warning (10335): Unrecognized synthesis attribute "src" at top.v(55) File: //wsl$/Ubuntu-18.04/home/lofty/de_10/build/top.v Line: 55
Warning (10335): Unrecognized synthesis attribute "init" at top.v(57) File: //wsl$/Ubuntu-18.04/home/lofty/de_10/build/top.v Line: 57
```

This is an example of the kind of spam that Quartus spits out when I try building any nMigen code. It also spits out warnings such as
```
Warning (10270): Verilog HDL Case Statement warning at top.v(1878): incomplete case statement has no default case item File: //wsl$/Ubuntu-18.04/home/lofty/de_10/build/top.v Line: 1878
Warning (10935): Verilog HDL Casex/Casez warning at top.v(1907): casex/casez item expression overlaps with a previous casex/casez item expression File: //wsl$/Ubuntu-18.04/home/lofty/de_10/build/top.v Line: 1907
```

But I think those are/should be separate issues.

whitequark, 2019-09-24:
In iCECube2 with LSE, all of our custom attributes appear to be ignored, or at least there are no complaints. In iCECube2 with Synplify, Synplify complains about all of our custom attributes. So I think they should be stripped in both cases.

whitequark, 2019-09-24:
Quartus, as @ZirconiumX shows above, complains about all of our custom attributes. So they should again be stripped.

> ```
> Warning (10270): Verilog HDL Case Statement warning at top.v(1878): incomplete case statement has no default case item File: //wsl$/Ubuntu-18.04/home/lofty/de_10/build/top.v Line: 1878
> Warning [...]

That's not actually an issue, just Quartus being obnoxious.

whitequark, 2019-09-24:
After 53bb4300a36a8e5a2f9f9613d484d75a73643273 I strip all internal attributes except `init` for `top.v`, and leave them all in `top.debug.v`, consistently for all platforms. `init` still has to be handled upstream. `src` could or could not be handled upstream as well, one day.

whitequark, 2019-10-12:
I've decided against stripping `(*init*)` attributes in nMigen itself using regular expressions, and rather waiting for a proper implementation to land in Yosys. Therefore, nothing is to be done here for 0.1.

whitequark, 2019-10-27:
Fixed in Yosys master.

closed 2019-10-27 by whitequark