m-labs/nmigen #225: Reconsider Signal.range() and Signal.enum()

whitequark, 2019-09-22:
We currently have three ways to specify the shape of a signal using `Record`/`Layout`: `int`, `(int, bool)`, `EnumSubclass`.

We currently have four ways to specify the shape of a signal using `Signal`: `int`, `(int, bool)` (both with the `Signal()` constructor), `EnumSubclass` (with the `Signal.enum()` constructor), `range` (with the `Signal.range()` constructor).

I think that in light of this, introducing separate `enum` and `range` constructors might have been a mistake. It should be possible to unambiguously fold these back into `Signal`. The initial decision was made because having both `shape` and `min`/`max` is ambiguous, but dispatching over the type of `shape` is not ambiguous and covers all of these use cases well.

emilazy, 2019-09-22:
I like the idea of `Signal(8), `Signal(range(..., ...))` and `Signal(MyEnum)`, but I really dislike `Signal((16, True))` and would only prefer this reform if a better alternative for that can be found (otherwise I'd rather have a `Signal.signed` and move it all to constructors).

whitequark, 2019-09-22:
I also dislike `Signal((16, True))`. It is OK that `Signal()` accepts this tuple, since this is what `Value.shape()` returns, and having `Signal(value.shape())` be valid is important, but it is a bad user interface.

I personally think having `Signal(8)`, `Signal(range(..., ...))`, `Signal(MyEnum)` and `Signal.signed(8)`, with no other constructors, is a reasonable compromise, since adding `signed()` to prelude for this single use case is not acceptable.

RobertBaruch, 2019-09-22:
`Signal(-8)` == `Signal.signed(8)`? :D

emilazy, 2019-09-22:
(here we go...)

whitequark, 2019-09-22:
@RobertBaruch The problem is that eventually someone tries to make a signal one bit shorter than x bits, and writes `Signal(x-1)`, and if x is zero, this does something totally unexpected instead of erroring out. So I don't think that's acceptable.

RobertBaruch, 2019-09-22:
I personally like all constructors, all the time. It's more readable to me than trying to guess what __init__ does.

whitequark, 2019-09-22:
That's reasonable, but what do we do with `Record`?

emilazy, 2019-09-22:
Records should be described with dictionaries rather than opaque tuples anyway.

whitequark, 2019-09-22:
That's not making it any better given that `(shape, direction)` is still a tuple.

commit 706bfaf5 (2019-10-11): hdl.ast: deprecate Signal.{range,enum}.

Although constructor methods can improve clarity, there are many
contexts in which it is useful to use range() as a shape: notably
Layout, but also Const and AnyConst/AnyValue. Instead of duplicating
these constructor methods everywhere (which is not even easily
possible for Layout), use casting to Shape, introduced in 6aabdc0a.

Fixes #225.
closed 2019-10-11 by whitequark