m-labs/nmigen #237: Signals should support (sig1 (not) in sig2)

emilazy, 2019-09-28:
This is for consistency with strings; Signals operate like strings (i.e. `sig[0]` is another Signal like with strings rather than element type like most collections, they can be sliced, etc.), so for consistency `__elem__` should support finding longer substrings than the one-bit case. (The windowing comparisons are also a little annoying to write out by hand, so it would be handy to have around, even if it's not necessarily a terribly common operation.)

whitequark, 2019-10-13:
This is actually completely impossible because [Python is bad](https://stackoverflow.com/questions/53351829/is-it-possible-for-contains-to-return-non-boolean-value).

closed 2019-10-13 by whitequark

mithro, 2019-10-13:
Maybe I don't understand what @emilazy was asking, but it seems like `sig in sig2` and `sig not in sig2` should work even when sig is multiple bits wide?

whitequark, 2019-10-13:
@mithro It's impossible to make the `in` operator return an nMigen signal (anything that isn't a boolean or convertible to it).