m-labs/nmigen #239: Indexing into a 0-length array

emilazy, 2019-09-28:
What does it *mean??* If nothing, then it should at least give a less messy error than it currently does.

whitequark, 2019-10-02:
My view on semantics is that since we allow things like 0-bit signals (at quite some effort), we should allow 0-length arrays, too. The shape of a proxy into a 0-length array is always `(0,False)`, and during legalization, it is treated like `Cat()` on both LHS and RHS.

Note that we cannot simply elide all code that refers to a 0-length array during legalization. For example, one could do something like `Cat(empty_array[index_signal], lhs_signal).eq(1)` and this would still need to assign `lhs_signal`.

An alternative would be to forbid creating a proxy into a 0-length array.

whitequark, 2019-10-06:
Turns out that's the semantics everything already implements, with the exception of a bug in RTLIL backend.

commit 2512a9a1 (2019-10-06): back.rtlil: don't crash legalizing values with no branches.

Fixes #239.
closed 2019-10-06 by whitequark