m-labs/nmigen #242: bitarray dependency is unfortunate

emilazy, 2019-09-30:
It's currently broken in PyPy and I feel like I've managed to poke it into memory unsafe behaviour from the CPython REPL before. Only pysim uses it so I imagine it could pretty easily be replaced with a combination of Python's bigints and lists.

emilazy, 2019-09-30:
Maybe Glasgow's `glasgow.support.bits` should be upstreamed and used instead?

whitequark, 2019-09-30:
I'm not going to touch pysim until 0.1, and there's a complete rewrite planned for 0.2 that will remove the dependency on `bitarray`.

Note that Glasgow depends on both anyway. `glasgow.support.bits` is immutable.

emilazy, 2019-09-30:
Glasgow only uses `bitarray` in a few applets, so I figured it could probably be migrated away.

whitequark, 2019-09-30:
I already tried to do that during migration to `bits`. I eventually decided to keep `bitarray` for mutable cases, though it might be possible to use a list of chunks anyway, assuming someone benchmarks it and figures out a good chunk size.

emilazy, 2019-09-30:
Fair enough; but in any case, using Glasgow on PyPy is a lot more niche than using nmigen on PyPy, and it has a lot more dependencies regardless.

xobs, 2019-10-01:
I thought I'd take a look at nmigen (migrating from migen), and I just ran into this issue.

```
❯ python .\haddecks.py --document-only
lxbuildenv: v2019.8.19.1 (run .\haddecks.py --lx-help for help)                                                                                                                                                            Traceback (most recent call last):
  File ".\haddecks.py", line 17, in <module>
    from litex.build.lattice import LatticePlatform
  File "D:\Code\had\haddecks\deps\litex\litex\__init__.py", line 4, in <module>
    from litex.soc.interconnect import packet
  File "D:\Code\had\haddecks\deps\litex\litex\soc\interconnect\packet.py", line 4, in <module>
    from nmigen.compat import *
  File "D:\Code\had\haddecks\deps\nmigen\nmigen\compat\__init__.py", line 8, in <module>
    from .sim import *
  File "D:\Code\had\haddecks\deps\nmigen\nmigen\compat\sim\__init__.py", line 5, in <module>
    from ...back.pysim import *
  File "D:\Code\had\haddecks\deps\nmigen\nmigen\back\pysim.py", line 5, in <module>
    from bitarray import bitarray
ModuleNotFoundError: No module named 'bitarray'
❯
```

I'm not actually using the simulator, so this dependency is definitely unfortunate.

whitequark, 2019-10-01:
@xobs Have you actually installed nMigen's dependencies? Because there is no goal to reduce the amount of dependencies to zero.

xobs, 2019-10-02:
@whitequark I've cloned the repository next to the existing `migen` repository, and changed my references to point there.

I was hoping `nmigen` would be a drop-in replacement for `migen`, however the addition of an installation step means that I will stick to `migen` for now.

mithro, 2019-10-02:
@xobs - You install `migen`, why wouldn't you install `nmigen`?

whitequark, 2019-10-02:
@xobs A user of nMigen is expected to be familiar with the absolute basics of Python package management. That's like saying "Yosys has dependencies, so I will stick to Vivado for now". I mean, sure, you can do that, but that's not a problem with Yosys or nMigen, it is a problem with your lack of desire to learn the fundamentals of the programming toolkit you are using.

xobs, 2019-10-02:
@mithro I don't install migen, I check it out.  I don't like using `pip` because it always breaks in spectacular ways.  It's difficult to run a workshop offline, and isn't included in all of the official python installations.  Running `pip install` has about a 50% success rate in my experience, so I very much like to avoid using it.

For example:

```cmd
C:\Users\smcro>python -m pip install git+https://github.com/m-labs/nmigen.git
Collecting git+https://github.com/m-labs/nmigen.git
  Cloning https://github.com/m-labs/nmigen.git to c:\users\smcro\appdata\local\temp\pip-y_btvmjq-build
Unknown requires Python '~=3.6' but the running Python is 3.5.2
You are using pip version 9.0.1, however version 19.2.3 is available.
You should consider upgrading via the 'python -m pip install --upgrade pip' command.

C:\Users\smcro>
```

With `migen` I just add it to my PYTHONPATH variable and go.  No need to mess with installing anything.  Plus it's much easier to make contributions, since it's just in a subdirectory.

cr1901, 2019-10-02:
@xobs nMigen requires 3.6 and above, though I'm acutely aware that upgrading Python and pip3 on Windoze can be a painful process.

whitequark, 2019-10-02:
@xobs In your example pip correctly tells you that nMigen requires Python 3.6, which is written in its README file:

> nMigen is designed for Python 3.6 and newer. nMigen's Verilog backend requires Yosys 0.9 or a newer version.

xobs, 2019-10-02:
I also have the official 3.7.3 installed:

```cmd
C:\Users\smcro>python --version
Python 3.7.3

C:\Users\smcro>python -m ensurepip --default-pip
D:\Software\ecp5-toolchain-windows-v1.5\bin\python.exe: No module named ensurepip

C:\Users\smcro>python -m pip install git+https://github.com/m-labs/nmigen.git
D:\Software\ecp5-toolchain-windows-v1.5\bin\python.exe: No module named pip

C:\Users\smcro>
```

whitequark, 2019-10-02:
OK, so what you're *actually* asking for is reliable installation instructions for Windows. I agree that this is something that is currently not provided but should be (especially given that you will also need Yosys to do anything).

plaes, 2019-10-02:
> I also have the official 3.7.3 installed:
> 
> ```batchfile
> C:\Users\smcro>python --version
> Python 3.7.3
> 
> C:\Users\smcro>python -m ensurepip --default-pip
> D:\Software\ecp5-toolchain-windows-v1.5\bin\python.exe: No module named ensurepip
> 
> C:\Users\smcro>python -m pip install git+https://github.com/m-labs/nmigen.git
> D:\Software\ecp5-toolchain-windows-v1.5\bin\python.exe: No module named pip
> 
> C:\Users\smcro>
> ```

Since 3.3 Python has [venv](https://docs.python.org/3/library/venv.html) module which allows one to have different application-specific package installs.

programmerjake, 2019-10-02:
maybe the windows installation issue should be in a separate issue

whitequark, 2019-10-03:
@xobs I've opened https://github.com/m-labs/nmigen/issues/244 to track the Windows installation problems.

mithro, 2019-10-03:
Just FYI - I've worked with @xobs a bit to give him some more Python background knowledge and some probably terrible advice ;-)

closed 2019-11-28 by whitequark

whitequark, 2019-11-28:
@xobs FYI as of today, nMigen master is pure Python. You only need to install `pyvcd` and only if you use the simulator. (Actually, it could probably be made an optional dependency even for that case, though the utility of a simulator that can't write waveforms is marginal.)

whitequark, 2020-07-01:
@xobs I apologize for being dismissive earlier. I now have a better understanding of the reasons someone would want to put packages on `PYTHONPATH` or copy them into `site-packages` directly without using `pip`. Although not a recommended way to use nMigen, I've made sure that the next release will support it properly, and I'll keep that working for foreseeable future. I also wrote detailed installation instructions, including for Windows: https://nmigen.info/nmigen/latest/install.html.

Please let me know if these changes work for you.

xobs, 2020-07-01:
@whitequark Thank you for the changes. I will take another look at nmigen. The approach is really exciting, so I'm looking forward to being able to try it out.

commit 7df70059 (2019-11-28): back.pysim: redesign the simulator.

The redesign introduces no fundamental incompatibilities, but it does
involve minor breaking changes:
  * The simulator commands were moved from hdl.ast to back.pysim
    (instead of only being reexported from back.pysim).
  * back.pysim.DeadlineError was removed.

Summary of changes:
  * The new simulator compiles HDL to Python code and is >6x faster.
    (The old one compiled HDL to lots of Python lambdas.)
  * The new simulator is a straightforward, rigorous implementation
    of the Synchronous Reactive Programming paradigm, instead of
    a pile of ad- [... cut]