m-labs/nmigen #245: Unused signal seems not to change

RobertBaruch, 2019-10-06:
Versions:

```
$ yosys -V
Yosys 0.9+932 (git sha1 10d0bad6, clang 6.0.0-1ubuntu2 -fPIC -Os)

$ pip3 show nmigen
Version: 0.1.dev700+g964c674
```

In this file, the signal `unf` is required to toggle on the negative clock, but doesn't, unless it is listed in the ports when calling `main`.

`test_case.py`:

```python
from nmigen import *
from nmigen.cli import main
from nmigen.asserts import *

class TestCase(Elaboratable):
    def __init__(self):
        self.unf = Signal()
        self.value = Signal(4)

    def elaborate(self, platform):
        m = Module()
        m.d.pos += self.value.eq(self.value + 1)
        m.d.neg += self.unf.eq(~self.unf)
        return m

if __name__ == "__main__":
    clk = Signal()
    rst = Signal()

    pos = ClockDomain()
    pos.clk = clk
    pos.rst = rst

    neg = ClockDomain(clk_edge="neg")
    neg.clk = clk
    neg.rst = rst

    testcase = TestCase()

    m = Module()
    m.domains.pos = pos
    m.domains.neg = neg
    m.submodules.testcase = testcase

    count = Signal.range(0, 61, reset_less=True)

    with m.If(count < 60):
        m.d.pos += count.eq(count + 1)

    m.d.comb += Assume(rst == (count < 4))

    m.d.comb += Cover(testcase.value == 5)

    #    main(m, ports=[clk, rst, testcase.unf, testcase.value])
    main(m, ports=[clk, rst, testcase.value])
```

`test_case.sby`:
```
[tasks]
cover

[options]
cover: mode cover
cover: depth 61
multiclock on

[engines]
smtbmc boolector

[script]
read_ilang test_case.il
prep -top top

[files]
test_case.il
```

Running:

```
$ python3 test_case.py generate -t il > test_case.il
$ sby -f test_case.sby
$ gtkwave test_case_cover/engine_0/trace0.vcd
```

When `unf` is not in the list of ports:

[image]

When `unf` is in the list of ports:

[image]

This is very inconvenient, especially when you get down multiple levels of modules and some of the signals in the modules are for testing purposes, and need to be seen without having to port them out.

whitequark, 2019-10-06:
Yes. The RTLIL backend has intentionally never made any guarantees about the behavior of unused signals. But maybe that should be changed.

whitequark, 2019-10-07:
Actually, this case is a bit special: the signal is both read and written, so it's not really unused (it depends on itself). I am not sure why this happens.

whitequark, 2019-10-07:
I believe this is not an nMigen issue. As you can see, the RTLIL it emits contains your signal (edited for clarity):
```
(* \nmigen.hierarchy  = "top.testcase" *)
module testcase(value, clk, rst);
  (* src = "t.py:14" *)
  wire \$4 ;
  (* src = "t.py:19" *)
  input clk;
  (* src = "t.py:20" *)
  input rst;
  (* init = 1'h0 *)
  (* src = "t.py:8" *)
  reg unf = 1'h0;
  (* src = "t.py:8" *)
  reg \unf$next ;
  assign \$4  = ~ (* src = "t.py:14" *) unf;
  always @(negedge clk)
      unf <= \unf$next ;
  always @* begin
    \unf$next  = \$4 ;
    (* src = "/home/whitequark/Projects/nmigen/nmigen/hdl/xfrm.py:528" *)
    casez (rst)
      1'h1:
          \unf$next  = 1'h0;
    endcase
  end
endmodule
```
Can you please confirm, and raise the issue upstream?

RobertBaruch, 2019-10-08:
Submitted [yosys issue 1441](https://github.com/YosysHQ/yosys/issues/1441)

whitequark, 2019-10-08:
Thanks. Closing as purely an upstream issue; will reopen if it turns out nMigen is involved.

closed 2019-10-08 by whitequark